motioneye-project/motioneyeos #824: Request: FTP upload scheduled and motion images

m4rky-m4rk, 2017-03-13:
Moved from motioneye to motioneyeOS

Hi,

I would like to FTP upload images to my website . Both on a schedule at regular intervals and also extra uploads should any motion occur. I already do this with my other IP cameras and would like MotionEye to do this too.

This very convenient. I just check my web page and see all images from all cameras. Usually its just scheduled half hourly shots which means all is running OK.

If FTP upload is not possible could MotionEye save images to a drive on a schedule AND also on motion. It seems that I cannot do both.

Thanks, Mark

maxholgasson commented 11 hours ago
Oh that would be great!!!
I spent now the entire evening trying to upload images taken by interval to a ftp server without success.. would be great

Oberlandix, 2017-04-06:
Hi there,
just rode your thread - I think it´s the same thing I am searching for:

I must install one more motionEYEos-Cam in a house. The problem is, that it´s in italy and there is just a connection with LTE to an "stupid" router (I don´t know the modell here). I am afraid, that the router will not allow to establish a connection with dynamic-dns service or sent me "every time if changed" the new IP-Adress.... It´s bad because so I wouldn´t be able to administrate the software from at home again.. There is still the hope that it will run....

At any way, there is just the way which is descriped above... To send every X hours a photo to a google-Drive... Isn´t it? Very nice would be the possibility to send a photo if there is a movement/action in the moinitored area (view of the cam). Would it be possible to send f.e. every day a photo, taken at 8 am, 12 am and 6 pm as EMAIL to a dedicated adress?

Sorry for my english.... But I think you can understand, whats my requirements to the cam...

Thank you very much, Oberlandix

m4rky-m4rk, 2017-04-06:
I have also LTE connection, from Drei, 3. My Drei router has very limited options but it can do DDNS. Even so, I have  set up second router installed behind the first. It works great. I have DHCP running on the second router only and on the first I turned it off and manually set an IP address to the second router. If you do the same then on the second router you could set up DDNS. There is probably a way to do it on a PI.

Oberlandix, 2017-04-06:
Dear m4rky-m4rk,
thank you very much. I´ll try to do the same you described -but I am afraid of the italian-router ;-) I HOPE that the router is able to do ddns, too.

Otherwise I must try it like described... sending a photo every X hours to an g-drive.

Greetings, Oberlandix

m4rky-m4rk, 2017-04-06:
As I said you are not stuck with the provided  ISP LTE router. You can add your own. I have a crappy  Austrian Drei router and a much better ASUS router behind that.

For the images you either save to a folder a regular snapshot based on a schedule OR if Motion occurs. You can not set up MotionEye to do both tasks. The folder is then synchronised to the cloud service. 

A friend of mine is working on creating a 'new option' to do both tasks, Motion images and Scheduled snapshots. Its part done. Just a config change but the web interface is taking a bit more time to show the new option correctly. When its done I could share the changes? I hope that's OK with CCrisan?

closed 2017-06-01 by bortek