motioneye-project/motioneyeos #830: Pi Zero W

bazzly1, 2017-03-16:
First off, this is very cool software! Thank you!!!
quick question or two.. Is motioneyeos working on the pi0-w? I had to plug a usb lan into it to get it up and running. Cant get the wireless running at all. Also does the camera module v2 work? It doesnt seem to see it. I'm using the Raspberry Pi NoIR Camera Module V2 - 8MP 1080P30.
Please let me know. 
Thanks

spieiga, 2017-03-16:
@bazzly1 Are you running motioneyeos 20170212 release?  If so, I believe you need to update the rpi firmware.

bazzly1, 2017-03-16:
@spieiga  Yes I am
Ahhh...didnt even think of that. I'll give that a go and see if that works. Thanks!

paulgant, 2017-03-17:
I agree, it's a great software product. I've had it working perfectly on a Pi2 with wifi dongle but, like you, I can't get it working on the ZeroW with either the v2 camera module or the on-board wifi. But I can get wifi working (although still not the camera) if I use a USB wifi dongle on the ZeroW rather than the on-board wifi. It's frustrating because MotioneyeOS and the ZeroW look, on paper, like the perfect partnership.

bazzly1, 2017-03-17:
So I did the update....Still a no go. So I'm going to assume its not compatible..?

wdawson001, 2017-03-17:
Follow the instructions here https://github.com/ccrisan/motioneyeos/issues/804#issuecomment-286071303

The official releases haven't been updated but I recompiled from the latest repo, so it works.

paulgant, 2017-03-18:
Excellent. Many thanks. It's working perfectly for me with on-board wifi, CSI camera, USB camera, and both together.

bazzly1, 2017-03-18:
Sweet! Thanks!!!

dragula96, 2017-03-18:
@wdawson001 thanks, to configure wifi i had to insert the imaged SD card to a rp1 using a cabled internet connection , then i connected to motion eye via web browser ip and configured the wireless using the gui, then re inserted the sd card back into the rp0w and it works like a charm (wireless + camera module v2.0).

fideocam, 2017-03-23:
Would it be possible to just configure /etc/network/interfaces and /etc/wpa_supplicant/wpa_supplicant.conf in the image file before burning img so there would not be any need to get a cabled internet connection?

wdawson001, 2017-03-23:
@fideocam absolutely, just prepare a wpa_supplicant.conf as demonstrated [here](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting-Raspberry-PI-Models-A-(Plus)-And-Zero) for a wireless install

Jay-Arr, 2017-03-23:
Getting a build error relating to the Zero W.  Started earlier today.

This is the error:

make[3]: *** No rule to make target `arch/arm/boot/dts/bcm2708-rpi-0-w.dtb'.  Stop.
  DTC     arch/arm/boot/dts/bcm2708-rpi-b.dtb
  DTC     arch/arm/boot/dts/bcm2708-rpi-b-plus.dtb
make[2]: *** [bcm2708-rpi-0-w.dtb] Error 2
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/home/sysop/motioneyeos/output/raspberrypi/build/linux-custom/.stamp_built] Error 2
make: *** [_all] Error 2

wdawson001, 2017-03-23:
@jay-arr I can compile it, but it doesn't load on the zero w - keeps getting a boot error. A branch has been merged and that always leads to some issues. Will try on my pi3 and report back.

Jay-Arr, 2017-03-23:
I'm trying to compile it on a system with WSL - I'm not saying that's a perfect build machine, but the error seems to indicate that it's only related to these recent changes.

wdawson001, 2017-03-23:
Yeah, not a great work flow. I use ubuntu under hyper-v. Works best for me. I can send you the lastest build but I haven't managed to boot it.

Jay-Arr, 2017-03-23:
I'll try Hyper-V too.  I'll see what I get for a result and report back (give me a day or so for the setup and compile).

Jay-Arr, 2017-03-23:
I'm going to try Ubuntu Server 16.10.  Why bother with a GUI, right?

wdawson001, 2017-03-23:
I went with the desktop version. Let me know how you get on.

wdawson001, 2017-03-23:
ah, it may just be to do with the buildroot-cache. Try clearing that out - in the motioneyeos folder .buildroot-ccache-raspberrypi

Jay-Arr, 2017-03-24:
Got it running now.  It's compiling - I guess...but it's throwing all kinds of syntax errors because of C++11.
Will wait until it's done to see if the result is a big steaming pile of bits.

wdawson001, 2017-03-24:
@Jay-Arr I tried to recompile last night, and it errors on No rule to make target arch/arm/boot/dts/bcm2708-rpi-0-w.dtb 

The Pi3 build went fine, so I think its missing a setting for the 0

Jay-Arr, 2017-03-24:
Tried it again.  Same error as I posted above.

Jay-Arr, 2017-03-24:
It has an extra line about the recipe for the pi zero dtb failing.

ccrisan, 2017-03-24:
Guys, I'm working on it. It seems that the latest kernel commit used by the current Raspbian distro does not have the DTS for the Zero W. They must have compiled the DTB using a newer version and provided it separately. I'll come up with a solution asap.

Jay-Arr, 2017-03-24:
No worries.  Semi-related: would installing motioneye (not OS) on an updated Raspbian Jessie Lite work for the time being?

ccrisan, 2017-03-24:
https://github.com/ccrisan/motioneyeos/commit/02b14d4c65b9688c29a788f12cc4ade6a29c00dd updates the kernel version so that it includes the required DTS. You'll need to (untested):

    git pull
    rm output/raspberrypi/.config
    ./build.sh raspberrypi linux-dirclean all
    ./build.sh raspberrypi mkrelease
    echo "Enjoy the Zero W!"

ccrisan, 2017-03-24:
@Jay-Arr yes, Raspbian appears to be working just fine on the Zero W and it should also get along well with motionEye.

Jay-Arr, 2017-03-24:
Ok.  Thanks for the updates.  You rock!

MattScherbatsky, 2017-03-25:
I've compiled it for my Raspberry Pi Zero W. You can download it at this link on my Google drive: https://drive.google.com/open?id=0B-DmMZQ20ALOWi1lM0FyOWJua0k

Thank you @ccrisan for your work!

toymanict, 2017-03-26:
Using MotionEyeOS I I can access the software through the browser. 

I am unable to see any display on HDMI, see the local camera or connect with a terminal.

Using motioneyeos-raspberrypi-20170307.img

fideocam, 2017-03-27:
Which img version should I be using with Zero W? There is one for pi 1, 2 and 3 (motioneyeos-raspberrypi3-20170212.img, motioneyeos-raspberrypi2-20170212.img.gz and motioneyeos-raspberrypi-20170212.img.gz).

wdawson001, 2017-03-27:
@fideocam motioneyeos-raspberrypi-20170326.img.gz

It's the one for the Pi1

ccrisan, 2017-03-27:
As @wdawson001 pointed out, the latest (pre)release supports the Zero W. The issue can be closed now.

closed 2017-03-27 by ccrisan

toymanict, 2017-03-27:
I have the Zero W, I used the motioneyeos-raspberrypi3-20170212.img version.
I even downloaded and tried the  motioneyeos-raspberrypi-201703##.img.gz
one. No luck.

I gave up on motioneyeos for my first one.

I followed a longer process by installing Jessie on a Pi 3, then RPi Zero
to get the software to work. Moved the SD to the Zero W and now it is
working.

Needed to get something working and finished.

Glad to help work out the bugs with motioneyeos on my next Zero W.

supremeliter, 2017-06-29:
MotionEyeOS was working perfectly on my raspberry pi zero 1.3.  I moved it to a pi zero w and now the frame rate is at nearly 0. Is this the same problem?

Also, respect to the dev. Great program and if I can help any way let me know.