motioneye-project/motioneyeos #832: Add instructions to the wiki for turning off the ACT LED on RPi Zero

per1234, 2017-03-17:
Unlike the other RPi boards, the Raspberry Pi Zero's ACT LED is active low so following the instructions at https://github.com/ccrisan/motioneyeos/wiki/Turn-Off-LEDs causes the ACT LED on the RPi Zero to be permanently on. I suggest adding the following text to that page:

For Raspberry Pi Zero:
```
# Disable the ACT LED.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=on
```

While I'm on the subject of that page I'd also like to suggest that if #184 is going to be mentioned there should be a link to that issue, which can be accomplished with the following markdown:
```markdown
As mentioned in [Issue #184](https://github.com/ccrisan/motioneyeos/issues/184).
```
renders as:
As mentioned in [Issue #184](https://github.com/ccrisan/motioneyeos/issues/184).

bortek, 2017-06-01:
thanks for suggestion.  Wiki page updated.

closed 2017-06-01 by bortek

Jachimo, 2018-04-25:
Perhaps worth noting that I don't think the suggestions in the Wiki work on ThingOS currently?  I don't think it has a `/boot/config.txt` that is easily editable.  The same thing can be accomplished (if you don't mind the lights flashing momentarily on boot) by putting a couple of lines into `/data/etc/userinit.sh` instead. Perhaps not as thorough a solution...

jasaw, 2018-04-27:
@Jachimo `boot` partition is mounted read-only by default. To remount it as read-write, run this command `mount -o remount,rw,noatime,sync /boot` and you should be able to edit `/boot/config.txt`.

desegura, 2022-01-31:
for pi zero 2 w run: `echo 0 | sudo tee /sys/class/leds/led0/brightness`