motioneye-project/motioneyeos #834: raspberrypi: no preferences, no way to log in or save images

akkana, 2017-03-18:
I installed motioneye for raspberrypi (I've tried two different images, motioneyeos-raspberrypi-20161125.img.gz and motioneyeos-raspberrypi-20170212.img.gz). In both cases, it boots and the web page shows the camera image, but there doesn't seem to be any way to get it to save images or change any other preferences. The documentation doesn't say how to do this, but when I google I find pages referencing preferences. The preferences shown for these two builds only offer three or four options: Fit Frames Vertically, Frame Rate Dimmer, Resolution Dimmer, and sometimes Layout Rows.

The installation instructions say, for first boot, "Click on the key icon on the upper-left side of the page to switch user to administrator. Use admin with no password when asked for credentials. Feel free to explore all the available configuration options." There is no key icon. Clicking on the person-torso icon in the upper left causes a reload, no login prompt (maybe it's automatically re-logging in since there's no password?) Clicking on the little owl in the upper right does the same thing.

Is there some extra step I need in order to make the preferences and login screen available? I can log in via ssh or the console, but I can't find anything in the installation instructions or the FAQ detailing what to do after first boot to enable the preferences and login screen.

ccrisan, 2017-03-18:
You're right about one thing: when I changed the icon from a "key" to a "user" I forgot to update this part of the doc. But that should be the way to log in as administrator. Have you tried different browsers? Have you tried clearing the browser cache?

akkana, 2017-03-18:
That was it. It was having a problem setting cookies. Now I can log in as admin, and can see some preferences. I still don't see all of them; e.g. in the Configuration wiki page it says there's a way to configure where it stores still images, and it also isn't showing a setting for camera resolution (it's defaulting to 320 x 200). Do things like that have to be configured via /etc/motioneye.conf?

ccrisan, 2017-03-18:
Just enable Advanced Settings in the Settings Panel.

closed 2017-03-18 by ccrisan

akkana, 2017-03-18:
With Advanced Settings, there are resolution settings under Video Device (but they don't work -- they require a reboot and after a reboot they've reverted to 640x480) but I still don't see anything about where still images are stored, and they're no longer accumulating in ~ftp/sdcard/Camera1/ (it saved 16 images there initially, then stopped) so I don't know if it's stopped collecting images, or just switched to storing them somewhere else.

Then after another attempt at changing resolution, after the automatic reboot it no longer saw the camera (I didn't touch the hardware) and I had to "click here to configure one". I'm getting a sense that maybe this isn't something I should even be trying on a base RPi (though my model B is listed under Supported Devices) and maybe should only be attempted on a Pi 3 with a second-gen camera model.

akkana, 2017-03-19:
Sure enough. I downloaded the pi 3 build and borrowed a pi 3, and the 3's web UI shows a "File storage" category that wasn't there on the basic pi build, as well as a few other new preference categories. So maybe the pi 1 isn't really supported any more even though it's on the supported list.

ccrisan, 2017-03-20:
All raspberry pi models will present the same features in motionEyeOS. You might have however enabled Fast Network Camera mode on your Pi1 and therefore some of the options might have been hidden.