motioneye-project/motioneyeos #835: Can't change the timezone

kanere23, 2017-03-18:
i have the motioneyeos recently installed and i can't change the timezone

my timezone if you are wondering is gmt+1.00
[image]

can someone help me with this problem.

wdawson001, 2017-03-18:
Change advanced settings to "on" - should find it then

Lama81, 2017-03-29:
There is a bug in the 20170326 release with timezone. When you change it in GUI, it stay at UTC after reboot.

ccrisan, 2017-03-29:
https://github.com/ccrisan/motioneyeos/commit/f57d036853660e38046882fe0ddddaec61244cc6 fixes the wrong date/timezone issue, although we might be talking about different issues.

closed 2017-03-29 by ccrisan

surak, 2017-04-06:
Where is the timezone supposed to appear? It's nowhere to be seen at the advanced settings. And it keeps saying that the most current release is from february...