motioneye-project/motioneyeos #836: smbus not installed on 20170212?

Snille, 2017-03-18:
Hi!
I've been using your excellent MotionEyeOS for about a month i 3 bird nests.
It's been working great!

But after the last update my "light.py" script can't find the smbus python module any longer.
In the /boot/config.txt I have added the following two lines:
```
dtparam=i2c1=on
dtparam=i2c_arm=on
```
In the /data/etc/modules I have these two lines:
```
i2c-bcm2708
i2c-dev
```
With lsmod I get:
```
...
i2c_dev                 5333  0
i2c_bcm2708             4943  0
...
```
So the modules are loaded...

Then in my script when I do: 
import sys, getopt, smbus

I get this error message when I run the script:
```
Traceback (most recent call last):
  File "/data/etc/light.py", line 26, in <module>
    import sys, getopt, smbus
ImportError: No module named smbus
```
You can find my script here: http://www.snille.net/download/light.zip
This did work on the last version of MotioEyeOS... 

What have I missed? 

I can't downgrade for the moment, I have a bird on one of my nests, don't want to disturb. :)

Best regards Erik

ccrisan, 2017-03-20:
I don't remember ever having added this package to motionEyeOS. You may want to use Raspbian + motionEye if you plan to do this kind of customizations.

closed 2017-03-20 by ccrisan

Snille, 2017-03-20:
Oh, hmm... It did work in the version before this one but maybe the smbus is the wrong way or not even needed? :)

What I really only need is the I2C interface to work, is that possible enabled some how?
My goal is simply to control this little thing: https://www.pi-supply.com/product/bright-pi-bright-white-ir-camera-light-raspberry-pi/

I really want to use your MotionOS because it's a read only system and when the appliance in this case is sitting in a bird nest in a tree outside. The risk of power loss is quite eminent. :)

bromhexine, 2017-03-29:
I've got the exact same issue with bright pi. Wish i had read this before i upgraded. Any way to downgrade back remotely?

ccrisan, 2017-03-29:
Downgrading is not (yet) possible. If this smbus is so important, I'll include it in the next builds. Honestly I had no idea it was part of the OS. It must have been pulled as a dependency for other package.

Snille, 2017-03-29:
Hi! I ended up modified versions of [these scripts](https://github.com/raspberrycoulis/brightpi-control) and scheduled with CRON instead. Works for me. :)

bromhexine, 2017-03-29:
thanks snille i'll check em out and try to copy what you've done. I havent upgraded in a long long time and i actually did make a full backup but stupidly deleted it before making sure the i2c was working. i do have a settings backup though but i dont think that'll restore all the things i had to do to originally get i2c working with the bright pi. ccrisan it will be greatly appreciated if the new update magically fixes it. Been using your work for years now with great satisfaction! (i probably shoulda just bought those infrared 2 led 3 watt with the auto light sensor on ebay for $15 or whatever but i2c seemed like a cool idea since it was customizeable).

ccrisan, 2017-03-29:
https://github.com/ccrisan/motioneyeos/commit/8a3638552c5c9901a73ae5550b607565ba6c0414 adds smbus package (available as of 20170329).

Gun-neR, 2018-10-13:
Hello.  I know this is an older topic... and that I should probably go the Raspbian + motionEye method... but I like the simplicity of using the motionEyeOS and just want to add basic control of my BrightPi... So going on the basis that this smbus has been added (whatever it is/does ;P ) I foraged ahead...  

I added the /data/etc/modules as showen above and did get this result with lsmod

```
i2c_dev                 6674  0
i2c_bcm2708             5741  0
```
which tells me nothing :P  except that it may "see" the i2c device?

Unfortunately all my limited existing knowledge on how to control the BrightPi seems nonfunctional in the Linux distro this uses.

For example... (also tried with sudo as well... even though already root)

```
[root@meye-847fb013 ~]# i2cset -y 1 0x70 0x00 0xa5
i2cset: write failed: Remote I/O error
```

@Snille I looked for your original script to work out how you did it, but the link is dead.  I even looked at your 2nd option... see above for one of the commands... but obviously do not understand what modifications you tried to make it work with motionEyeOS

Am I missing something simple?

Gun-neR, 2018-10-13:
OK... somehow I think I killed my Bright Pi and that is why it is giving the I/O error... as now it will not work in a known good scenario with another RPi.  and _i2cdetect -y 1_  just shows this on any RPi I try it on:

[root@meye-847fb013 ~]# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

Oh well :(

Snille, 2018-10-13:
@Gun-neR: Too bad your BrightPi died... However, I lust updated the link to the "new" scripts in the fist post. :)