motioneye-project/motioneyeos #843: Change Admin password (20170212)

Eska72, 2017-03-22:
When i change the Admin password during a clean install, the system crashes. So i installed an older version, 20161027, and then i have non problems when i change the admin password. Ater that i checked for updates and the newest version will be installed. Workaround but fixed.

breadfan13, 2017-03-22:
are you changing it from the command line

ccrisan, 2017-03-22:
It would help telling us how exactly does it crash? Log files, error messages & other debugging details could be useful.

Eska72, 2017-03-23:
I use MotionEyeOs latest version on a Raspi3. I changed the admin password on a webinterface, using the ipadress of the Raspi. After changing you need to reboot. Then the raspi is out of reach, the webinterface keeps searching, but can't find the raspi anymore. So i connect the raspi to a monitor to see what happens. Everything seems to work fine, but the operating system won't start. I have to use the admin login with the new password and then it stays ath the command prompt. Ls gives sdcard/storage.
After when i give the command reboot, i can reach the raspi again by the webinterface. I've really no idea what happened but it seems to work fine now. 

boot.log:
---- booting motionEyeOS 20170212----
 * Creating data partition: done
 * Formatting data partition: done
 * Mounting data partition: done
 * Creating required data files: done
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting watchdog: done
 * Starting eudev: done
 * Setting root password: done
 * Configuring wired network: dhcp
 * Setting current date using http: Sat Mar 18 14:15:45 UTC 2017
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Setting smb admin password: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.178.17/24
 # Default gateway is 192.168.178.1
 # DNS server address is *****
---- shutting down motionEyeOS 20170212 ----
 * Stopping motioneye: done
 * Stopping smbd: done
 * Stopping nmbd: done
 * Stopping proftpd: done
 * Stopping sshd: done
 * Stopping crond: done
 * Stopping date updater: done
 * Stopping network: done
 * Stopping eudev: done
 * Stopping syslogd: done
---- booting motionEyeOS 20170212----
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting watchdog: done
 * Starting eudev: done
 * Setting root password: done
 * Configuring wired network: dhcp
 * Setting current date using http: Sat Mar 18 15:07:50 UTC 2017
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Setting smb admin password: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.178.17/24
 # Default gateway is 192.168.178.1
 # DNS server address is ******
---- shutting down motionEyeOS 20170212 ----
 * Stopping motioneye: done
 * Stopping smbd: done
 * Stopping nmbd: done
 * Stopping proftpd: done
 * Stopping sshd: done
 * Stopping crond: done
[... 128 lines cut ...]
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting watchdog: done
 * Starting eudev: done
 * Setting root password: done
 * Configuring wired network: dhcp
 * Setting current date using http: Thu Mar 23 15:46:16 CET 2017
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Setting smb admin password: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.2.16/24
 # Default gateway is 192.168.2.1
 # DNS server address is 192.168.2.1

[messages.2log.txt](https://github.com/ccrisan/motioneyeos/files/865345/messages.2log.txt)

Attached a part of the copy of the messages.lof file. Tried to change it on 19 march, fixed it today 23 march. Dunno how...

gr

ccrisan, 2017-03-23:
I'm sorry but to me this works just fine. I see no problem in your logs.

closed 2017-06-01 by bortek