motioneye-project/motioneyeos #844: Support for Cubitruck

flwah, 2017-03-22:
Hi,
with version 20161212 you did support the cubitruck board, which I would like as video recording station for my Pi3s, due to the SATA support. Are you planning on supporting that boards in further versions, or are you focusing on PI and odroid?

Thanks!

flwah

ccrisan, 2017-03-22:
This is a good question. Normally I wanted to give up support for Cubietruck given the extremely few users (as shown by the download counters). The Cubietruck build is mostly identical to the Banana PI one. I'll try to make the Banana PI build compatible with the Cubietruck and support both boards under the same image file.

ccrisan, 2017-05-07:
I'm sorry but I cannot seem to make the ethernet work reliably on cubietruck. It works fine with Banana PI though. I don't plan to support Cubietruck anymore as it's already an old, relatively big board.

closed 2017-05-07 by ccrisan