motioneye-project/motioneyeos #848: Request:  Camera button in live stream UI to manually take a snapshot

Osirus126, 2017-03-24:
It would be great to have a button to manually take a picture via the web interface. If it is possible, that would be a useful feature. 
- You could have the "Capture Frame" button placed next to the "Toggle Full-Screen Camera" button.
- It would take a picture with the camera that is selected, or there could be a multiple selection option so you can select more than one camera at once.
- It could have a three frame capture option as well.

is this idea possible?? if so would it be hard to accomplish? Would anyone else like this option implemented?

I would love to just do it myself but dont know where to start. I am a novice at programming.

ccrisan, 2017-03-24:
It's something I've been meaning to implement for a long time. It's going to happen some time in the future but as far as I remember there were some problems with motion crashing when instructed to take a snapshot. Anyways I'll see what I can do.

Osirus126, 2017-03-24:
is the Web UI built with java? Maybe there is a way to capture the window using java in some way. 

http://www.javacreed.com/take-a-screenshot-of-web-page/

timdonovanuk, 2017-06-07:
Just came to request this too! +1

timdonovanuk, 2017-06-08:
This is super easy, just create an action button that executes this:

curl http://localhost/picture/1/current/ -o /data/output/Camera1/`date +%Y-%m-%d_%H-%M-%S`.jpg

This will save a snapshot of the current frame. No need to contact motion to tell it to take a snapshot.

Obviously if added to the main UI it  would need to read in the config settings for storage path and file naming.

timdonovanuk, 2017-06-08:
For other people, here is how to do this:

1) ssh into your pi
2) mount -o remount,rw /
3) nano /data/etc/preset1_1
4) Put in the following and save (change locations, camera number and date format if needed):

#!/bin/bash

curl http://localhost/picture/1/current/ -o /data/output/Camera1/`date +%Y-%m-%d_%H-%M-%S`.jpg

5) chmod +x /data/etc/preset1_1

Reload motioneye in your browser. You'll see a little (1) in the bottom  right hand corner when clicking on a camera.

@bortek just as a side note, all the docs say the config files (motioneye.conf etc.) and location of action buttons is in /etc/motioneye/ but it's actually in /data/etc. Cheers.

ccrisan, 2017-08-05:
If what you really need is just to tell motionEye to take a snapshot and save it as a regular still image, https://github.com/ccrisan/motioneye/commit/55035be4624bad97e8fd262aa5b91aa196c373e2 adds support for that. Otherwise, if you need access to the snapshot file to do other stuff, the solution above should work.

closed 2017-08-05 by ccrisan

Mega1024, 2021-10-18:
> Settings, Still images, Enable Manual Snapshots
> [image: image]
> <https://user-images.githubusercontent.com/20134429/137687935-1da1a132-1830-4807-8ff3-adc87ce15693.png>
> New camera button in lower right of video box
> [image: tempsnip]
> <https://user-images.githubusercontent.com/20134429/137688732-9e60cf6b-3e92-4e5a-9d76-0e610f2c90bd.png>

Hey thanks for the quick response - I actually found it 5 minutes after I
posted, and deleted my post :)