motioneye-project/motioneyeos #853: 0170326 breaks pi camera

snagglewest, 2017-03-28:
Update 20170326
Raspberry Pi 1 Model B
Raspberry Pi 5MP Camera Board

Camera does not to function in fast network camera mode, just the circle/slash symbol displayed in place of the picture.  Switching to normal mode the camera works.

ccrisan, 2017-03-28:
I'll look into it.

fredericcollin, 2017-03-28:
I confirm that updating to 20170326 does break the O.S. It doesn't even boot. However a new install from scratch does work as it should.

Monki77, 2017-03-28:
Same issue here with fast network cam not working.
PiZero W
Pi NOIR v2 cam

gridbox, 2017-03-28:
It appears the the `picamera` python module is missing in the new build.  

```
[root@meye-753ec24e ~]# raspimjpeg.py 
Traceback (most recent call last):
  File "/usr/bin/raspimjpeg.py", line 23, in <module>
    import picamera
ImportError: No module named picamera
```

gridbox, 2017-03-28:
As a quick fix, I added `BR2_PACKAGE_PYTHON_PICAMERA=y` to `raspberrypi_defconfig` and rebuilt from source.  Fast network camera mode now works.

fredericcollin, 2017-03-29:
I also noticed that the time is incorrect. It seems to display UTC time although I have set my timezone correctly.

ccrisan, 2017-03-29:
@fredericcollin what board are you using?

fredericcollin, 2017-03-29:
RPI2

Lama81, 2017-03-29:
Yes timezone is always at UTC now beside seting in web interface

ccrisan, 2017-03-29:
https://github.com/ccrisan/motioneyeos/commit/f8e22cbac8b9c16fb3054efa3df44bd079cf6dc5 fixes the FNC mode (thanks @gridbox for finding the missing package).

https://github.com/ccrisan/motioneyeos/commit/f57d036853660e38046882fe0ddddaec61244cc6 fixes the wrong date/timezone setting.

closed 2017-03-29 by ccrisan