motioneye-project/motioneyeos #857: Mac OSX - writeimage.sh string indexing error

Mitoflyboy, 2017-03-30:
Hi,
I am running OSX 10.12.3, and when I run the 'writeimage.sh' script I get the error:

**:: decompressing the gzipped image
./writeimage.sh: line 83: -3: substring expression < 0**

After some research it appears that the bash version on OSX doesnt support negative string indexing. The solution is below:

-    $gunzip -c $DISK_IMG > ${DISK_IMG::-3}
-    DISK_IMG=${DISK_IMG::-3}
+    $gunzip -c $DISK_IMG > ${DISK_IMG::$#-3}
+    DISK_IMG=${DISK_IMG::$#-3}

Thanks

thomaskuczek, 2017-03-30:
I was creating a motioneyeos sd card today using https://etcher.io/.
Etcher is also recommended by Raspberry Pi installation guide here: https://www.raspberrypi.org/documentation/installation/installing-images/mac.md

Maybe the documentation should be altered and also recommend etcher?

bortek, 2017-06-01:
thanks I have updated wiki page with this info 
https://github.com/ccrisan/motioneyeos/wiki/Installation

closed 2017-06-01 by bortek

Choscura, 2017-08-17:
I was having trouble running writeimage.sh on Fedora25, I was getting a few errors and I'm not sure if I'm doing something wrong or what.

sudo ./writeimage.sh -d /dev/sdb1 -i "motioneyeos-raspberrypi-20170710.img.gz" -n 'secret:secret'
[sudo] password for choscura: 
:: decompressing the gzipped image
./writeimage.sh: line 83: motioneyeos-raspberrypi-20170710.img: Is a directory