motioneye-project/motioneyeos #858: Credentials working for SSH but not for the web interface?

Flohhupper, 2017-03-30:
I think I go crazy. 
I changed the admin and the user password on the web interface and retarted my PI.
Now I am not able to login to either one. I am, however, able to log into the admin account via KiTTY with the credentials I set.

Am I doing something wrong? Is it possible to change the credentials over SSH?

ccrisan, 2017-03-30:
Have you set your admin password to something that contains special characters? If yes, please tell us what those would be so we can see if they pose a problem.

You can change your password via SSH by editing `/data/etc/motion.conf` and looking for `@admin_password`, then rebooting using `reboot`.

Flohhupper, 2017-03-30:
I did use special characters. The password contains # _ + ß (Hash, Underline, plus, German Eszett).
I thought to myself that those could be problamatic, but after I tried them in Kitty without problems, I thought that this could not be the problem.

Flohhupper, 2017-03-30:
Update: I changed the passwords in the file @ccrisan pointed at and it worked. I can't believe that I fell for that stupid mistake. Thanks for your help!

closed 2017-03-30 by Flohhupper