motioneye-project/motioneyeos #859: Dropbox: Accessing the upload service failed: HTTP Error 409: Conflict!

thistempname, 2017-03-30:
I've searched the history and nothing came up.

Folder is created on Dropbox.
Opened it up to the optional other users (0/500)
Cameras (2) are local (USB into RPi3)
Obtain Key
Test Service
Result Same for Both Cameras

...I receive the error message...
'Accessing the upload service failed: HTTP Error 409: Conflict!'

What should I be troubleshooting?

Thank you all in advance.

FlyingPersian, 2017-04-17:
I have the same error. I filled out everything, then hit apply, then hit test and got that error. 

motionEyeOS 20170329 on a Pi b+

ccrisan, 2017-04-19:
Try disabling the uploading service, reenabling it and refilling everything from scratch.

closed 2017-06-01 by bortek

laresbernardo, 2017-09-29:
Same here but generating links: `Conflict (HTTP 409)`
```
token <- drop_auth()
files <- drop_dir(path = "/Documentos/mypath/")
head(files) %>% data.frame()
drop_share(path = files$path_display[1],
           requested_visibility = "password",
           link_password = "mypass",
           dtoken = token)
```
Weirdest thing of all, only worked for me once and then it gave me the `Conflict (HTTP 409)` error!
I'm using the latest github version and tried with the CRAN's as well.. nothing.
Oh, I have a Premium account so the password thing should not be a problem.

outcomeproject, 2017-10-25:
Hi,

I had a working shiny app using rdrop2 for sometime.
When I tried to use it today the app did not retrieved the files that are within DropBox.

Back to R to study what happened I found that that drop_get is deprecated and that I received now the following error:

Downloading: 94 B     Warning: Error in drop_download: Conflict (HTTP 409).
Stack trace (innermost first):
    63: <Anonymous>
    62: stop
    61: httr::stop_for_status
    60: drop_download
    59: eval [/My/DropBox/File/Path/app.R#404]
    58: eval
    57: withProgress
    56: observerFunc [/My/DropBox/File/Path//app.R#399]
     1: runApp

I changed drop_get() to drop_download() and I have read posts on 409 error messages but I have not came up with a solution.
Help on this is really appreciated. Thanks

Jaizan, 2019-11-20:
I also have the same error "Accessing the upload service failed: HTTP Error 409: Conflict!".
I tried switching off the upload, then refilling everything from scratch, with a new drop box key.   No success.
I also tried rebooting the Pi.    

This pi had been working fine on Google until I swapped the camera and needed to set a new upload path for the new Camera.   Google has the known "Sign in with Google temporarily disabled for this app" which occurs any time I ask for a new key.   Existing google key works OK on another device.   So I have no available 3rd party storage, with Google refusing new API key requests and the HTTP Error 409 with Dropbox.

oliveox, 2019-12-20:
I could make it work.

Service: Dropbox
Location: /camera
Include subfolders: yes

Only switching, refilling didn't work for me. Not very scientific but I manually created in Dropbox the folder specified at "Location" and then did the switching --> refilling thing (hitting apply after blanking all options under "Upload Media Files" and switching it off ). 

Magically, it worked.

jshep321, 2020-08-05:
Hi ,
FWIW I was trying all the hard stuff and then I realized it was stupid user error...  I made the new directory in dropbox..... but I didn't have the full directory path in the motioneye UI.
securitycams/thisnewcamera instead of thisnewcamera did it for me.

monolyf, 2021-11-23:
I had this issue and realised I was using a backslash ( \ ) instead of forward slash ( / ) 

As windows explorer uses backslashes in its file directory and everywhere else seems to use forward slashes. All working now