motioneye-project/motioneyeos #861: How do I switch on the timelapse feature?

thomaskuczek, 2017-03-31:
Timelapse is mentioned as feature here:
https://github.com/ccrisan/motioneyeos/wiki/Features

But how to switch it on?

ccrisan, 2017-04-02:
You don't switch it on. You build it from the pictures already taken. Use the media browser for that.

closed 2017-04-02 by ccrisan

tommy-rk, 2017-04-02:
But motion has a feature to generate timelapse automatiacally.
See: http://www.lavrsen.dk/foswiki/bin/view/Motion/MotionGuideBasicFeatures

Timelapse has an own filename: Note that the options snapshot_filename,
jpeg_filename, ffmpeg_filename, and timelapse_filename all allow specifying
directories by using '/' in the filename.

It would be easy to use motions feature and build it into motioneye.

Thomas

ccrisan, 2017-04-02:
I know motion has its own timelapse feature and you're free to enable it manually (using Extra Motion Options). motionEye uses the method described above because of its flexibility (you can choose at "runtime" some parameters).

tommy-rk, 2017-04-02:
I like it, if motion is creating a timelapse video for everyday
automatically without me having to create it manually.
How do I need to configure Extra Motion Options in motioneye, so motion is
creating a timelapse video automatically for every day?
I am using motioneyeos (which rocks! thanks a lot!)

surak, 2018-08-03:
When you choose it on the media browser, it shows a progress bar, which completes and the screen is back to the pictures, and nothing happens. Where does the timelapse go to?

afeno, 2018-10-29:
I found some Extra Motion Options that seems to have the work done:
timelapse_fps 30
timelapse_mode daily
timelapse_interval 1

Now I need to check how to get a format that that be played from the browser directly from MotionEye. I don't want to download the file to watch it.

afeno, 2018-10-31:
Im trying to set "movie_codec mp4" in the Extra Motion Options but it doesn't take it. The file is still generated as .avi
Any idea why? How can I see the video files generated by Motion directly from the browser? I can't play the video files that are in the (ungrouped) category.

cinnamonsweets, 2019-02-05:
> Im trying to set "movie_codec mp4" in the Extra Motion Options but it doesn't take it. The file is still generated as .avi

https://motion-project.github.io/motion_config.html#timelapse_interval

> Note that different containers are NOT planned for this option since the nature of recording a timelapse is not what typical movies containers and codecs are designed to do. Users desiring a different container/codec can use ffmpeg directly to either re-encode a finished timelapse video or output the timelapse as jpg picture files and use ffmpeg to encode those jpgs into a movie.