motioneye-project/motioneyeos #863: Installation with writeimage.sh impossible os x and linux

cassman51, 2017-04-01:
Hello,
After several attempts to install I remain blocked. 
I tried installing motioeyeos of my osx but the message writeimage.sh command not found appeared despite the script's right change. I then try installing using raspbian, after several attempts I still arrive at the same issue. 
The installation this spear and in the final I have as answer: 
220+0 record read
220+0 record write
230686720 bytes (231 MB) copy, 53,145s, 4,3MB/s
:: re-readîg sdcard partition table 
Error: Partition(s) 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64 on dev/sda1 have been written, but we have been unable to inform the kernel of the change, probably because it/they arr in use. As a result, the old partition(s) will remain in use. You should reboot before making further changes. 
I am totally lost. I am novice in the world of linux I have already put two days to understand how to use the script writeimage.sh according to my hardware. For info the command that I type to start the installation is:
sudo /home/pi/Desktop/writeimage.sh -d /dev/sda1 -i /home/pi/Desktop/motioneyeos-raspberrypi-20170329.img -n "my ssid:my passwork"

I would like use motioneyeos in a raspberry pi zero w.
I thank you in advance for the time you will give me

ccrisan, 2017-04-01:
I think you could ignore those errors and try to boot the sd card.

cassman51, 2017-04-01:
When I boot with the card on my pi zero or zero w nothing happens as if no card is present

ccrisan, 2017-04-01:
Also you should use /dev/sda (the entire disk), not just the first partition (/dev/sda1).

cassman51, 2017-04-01:
I will try and keep you informed

cassman51, 2017-04-01:
it works ! thanks a lot for your help.

closed 2017-04-01 by cassman51