motioneye-project/motioneyeos #865: Raspberry Pi 1, 2 or 3?

thomaskuczek, 2017-04-01:
Which Raspberry Pi is the best when using the Raspberry Cam? Are there difference in frame rate?

SuneBielefeldt, 2017-04-01:
# **Best Raspberry Pi version to MotionEye**
As you can see on the picture below, there is a big difference in frame rate depending on the model you use. See frame rate by the red dots.

The camera I use, run in 1600 x 1200, except Cam6 which is a Logitech USB camera which can only handle 1024 x 786 pixels.

I use both Raspberry Pi 1.2 and 3, and version 3 provides at least twice the version first

I have no experience with P Zero W yet.

Version 3 is the best choice.

[image]

cephasiii, 2017-04-02:
What case do you use? Are your cameras located inside or outside?

jezmck, 2017-04-05:
I have an original Pi (B 1.2), a Zero and a ZeroW, none are stable/reliable at more than 1024x768.

jezmck, 2017-04-05:
@SuneBielefeldt What are the red dots on your camera feeds?

SuneBielefeldt, 2017-04-05:
Best Raspberry Pi version to MotionEye

As you can see on the picture below, there is a big difference in frame rate depending on the model you use. **See frame rate by the red dots.**

SuneBielefeldt, 2017-04-05:
All Cameras are placed inside

SuneBielefeldt, 2017-04-05:
The last two Camera's are placed Remote (25 km away) by a frend

closed 2017-06-01 by bortek