motioneye-project/motioneyeos #866: Camera Blur

benmeryon, 2017-04-01:
Hi !

I bought 2 Rpi Cams recently and I'm a little bit disapointed by the camera quality.
Here a a screenshot of the two cameras: http://imgur.com/a/WwUl4
Can you tell me if the reslut is normal?

Thanks a lot !

roycie1282, 2017-04-02:
I believe u need to focus ur cameras the new 8mp RPI cameras are focused on for roughly a 10 ft shot. Some of the more recent cameras have came with a focusing tool. But if u didn't get one you can drill an old credit card or something similar etch out 4 little notches inside the circle to basically make it a square and it will work as a focusing tool. I found a thread somewhere on Google how to do it and it worked without issue and really cleaned up the image.

Also if u have the lens pressed up against anything it will take it out of focus I bought some camera cases that kept pushing it back out of focus so I had to shave down the case so the camera would sit correctly. Hope this helps

silversword411, 2017-05-18:
Yep, check the manual focus (that's what that round white piece of plastic was for) :)

closed 2017-05-18 by silversword411