motioneye-project/motioneyeos #867: Price comparison - Raspberry pi vs. Foscam C2

tommy-rk, 2017-04-02:
This might be a stupid question, but I am wondering how Foscam C2 can be so cheap and still has great technical specs compared to a raspberry solution?

When I sum up all what I need to build a surveillance cam with raspberry pi I need about:
EUR 95,26 (amazon new price for all components needed.)

Foscam C2 cost 99.90 Euro and has 1080p fullHD resolution with about 25fps.

How can Foscam be so cheap? Am I comparing apples with oranges?

sgiannulli, 2017-04-03:
maybe this is not the right place where to make such considerations....just think of what you can do with Foscam C2 and what you can do with an RPi3 with raspbian jessie installed on top; more people complain about what Foscam writes as technical specifications and what their cams really do.

cdevelc, 2017-04-04:
I own a Foscam C2 and recently bought and configured a Raspberry pi + camera + motioneyeos.  First, I'm not sure why OP would be surprised about the Foscam price point.  From a hardware standpoint it is just about the same amount of tech as a Raspberry pI + camera.  The C2 is a microprocessor board + camera + case.  The real thing driving the price is probably the volume, not the actual tech.  A similar price point for the gear is really not surprising.

But, motioneyeos is a much, much better software platform, IMHO.  The Foscam software does not support motion detection through a window (PIR does not work through glass).  The Foscam software is closed source.  You are completely at the mercy of Foscam to add (or often remove) features.  I owned another Foscam model for many years.  It only worked on Firefox.  They never added Chrome support.

tommy-rk, 2017-04-04:
That is some really good feedback - thanks a lot.

bortek, 2017-06-01:
https://github.com/ccrisan/motioneyeos/wiki/Installation

closed 2017-06-01 by bortek