motioneye-project/motioneyeos #874: Timelapse camera's, not getting the maximum resolution.

sderoos, 2017-04-06:
Hello! 

First of all I want to say thank you for all the effort put in this project, it's awesome! 

We are using this OS on three Pi's that are creating a timelapse on a shipyard where a new ship is being build. Before I found this project we used some scripts that created images and uploads them to a webserver. Those scripts created images that had the max resolution (2592×1944). 

In Motioneye I can set this resolution, the images are 2592×1944, but it looks like the image is just enlarged. We have a much smaller view now with Motioneye.

 Example from same point of view with old timelapse scripts (sorry, I don't have original images, but you will get it):
http://nl.tinypic.com/r/10xxe9j/9

Motioneye when using normal resolution (1920x1080):
http://nl.tinypic.com/r/mv0z2t/9

Motioneye when resolution is set to 2592x1944:
http://nl.tinypic.com/r/i4pkqp/9

This is a small timelapse I made with MotionEyeOS with the outside camera:) :
https://vimeo.com/211845633

ccrisan, 2017-04-06:
Well the two resolutions don't have the same aspect ratio (1.77 vs 1.33). Instead of distorting the image, the camera firmware apparently prefers cutting it. It may also have to deal with video mode vs. stills mode which could be controlled via bcm2835-v4l2 module parameters. It needs some experimentation though. I recommend that you seek help on RPi-specific forums and get back with a solution, as I don't have one for you :)

sderoos, 2017-04-06:
Ok, thank you!

closed 2017-04-06 by sderoos