motioneye-project/motioneyeos #876: PI Zero W : unable to connect to WIFI

bre75, 2017-04-06:
Hi,

I have Motion Eye OS installed and working as a charm on a Raspberry PI Zero connected to my WIFI.

I bought a PI Zero W and installed the latest build of Motion Eye OS on it. Still to work with my WIFI.

It's not working for the following reason : Starting wpa_supplicant : failed (brcmfmac)" (see attached file)

The file is well there with the right information to connect to my WIFI;
To be sure the PI Zero W is working well i used another sd card with Raspbian and the WIFI is connecting without any problem.

Any idea what could be the issue ?

Thanks a lot for your help,

Best regards,
[image]

rc999985, 2017-04-07:
https://github.com/ccrisan/motioneyeos/issues/804#issuecomment-286071303

This worked for me and I had the same issue.

Good luck

rc999985, 2017-04-07:
Also looks like there's a pre release 

https://github.com/ccrisan/motioneyeos/releases/tag/20170326

ccrisan, 2017-04-07:
Have you checked this troubleshooting article: https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems?

bre75, 2017-04-07:
Hi,

Yes i checked everything.

Strange.
This morning I did 3 full install from scratch. 
Without changing anything in the way i installed (with writeimage.sh on my Ubuntu).
I used the exactly same command line and the same sd card every time (that i format between each installation).
At the 3rd install ... It worked and i succeed to connect to my wifi.

ccrisan, 2017-04-07:
@bre75 then it might be related to power issues.

bre75, 2017-04-07:
Power issues ?
From the Raspberry ?

RawBOT, 2017-04-07:
I think I found out what the problem to this is, as it happened to me. It appears that you need to setup wpa_supplicant.conf on the first boot. If for some reason your first boot doesn't include the wpa_supplicant.conf, MotionEyeOS defaults to Ethernet, which it cannot find. This makes every subsequent boot fail, as it never looks again for the WiFi config, and just reboots because Ethernet is unavailable.

I think the best solution would be to search for a WiFi config file everytime Ethernet is unavailable on boot.

This was tested on my Raspberry Pi Zero W.

rc999985, 2017-04-07:
Yes note steps 3 and 4 here https://github.com/ccrisan/motioneyeos/issues/804#issuecomment-286071303

bre75, 2017-04-07:
It's driving me crazy ..
Did a full fresh installation on an other new Raspberry Pi Zero W, with a new SD Card.
On a Windows computer this time.
And exactly the same issue.

ccrisan, 2017-04-07:
@bre75, have you tried replacing the power supply with a more powerful one with shorter or thicker cables?

bre75, 2017-04-07:
Ah yes i forgot to mention it.
It's also another an new power supply.
Official Raspberry one

RawBOT, 2017-04-08:
> On a Windows computer this time.

Make sure you don't use Notepad and use a text editor that lets you use Unix EOL (LF).

RPiSpy, 2017-04-08:
I used Notepad++ and made sure it was set to "Linux" style line breaks (it tells you in the status bar bottom right).

You have make sure the file is correct. Otherwise you may need to re-write the SD card and start from scratch.

Working fine on a Pi Zero W.

farangdude, 2017-05-18:
Has there been any progress on determining the cause of the wifi failure indicated in the original post (#876)?  I had the exact same failure using the most recent release motioneyeos-raspberrypi-20170329.  Including the same output as shown in the screenshot in the original post with the same error "Starting wpa_supplicant : failed (brcmfmac)".  This was from a fresh write to the SD card.  

I am curious to the significance of the "(brcmfmac)" portion of the error and what it is trying to indicate.

Thanks

efetzer, 2017-05-29:
I experienced all of the issues associated to "Starting wpa_supplicant : failed (brcmfmac)"...

Solved this way for me...
Immediately after running the install script...
Reinsert sdcard 
Open wpasupplicant.conf
Remove the blank lines after the } with a text editor....
Save...
Eject...
SUCCESS !!!

closed 2017-06-01 by bortek

plankton99, 2017-10-13:
I have tried everything above, still get the "starting wpa_supplicant: failed (brcmfmac)"
this is SO frustrating.  any one have a real solution to this?

ccrisan, 2017-10-13:
@plankton99 does your board's WiFi connection work well with Raspbian?

plankton99, 2017-10-13:
Yes both pi3 and zero w connect fine w raspbian

ccrisan, 2017-10-13:
@plankton99 do you use exactly the same wpa_supplicant.conf file in all the cases?

plankton99, 2017-10-13:
i used the raspbian pi jesse desktop and i can connect to the wifi via the desktop wifi set up.

ccrisan, 2017-10-14:
@plankton99 please try using the exact wpa_supplicant.conf file and see if it works. If it doesn't, try identifying the differences between Rasbpian's generated wpa_supplicant.conf and yours.

plankton99, 2017-10-14:
@ccrisan on the RASPBIAN STRETCH WITH DESKTOP which i can connect to the internet with on the PI3 board, in the root directory on the SD card there is no file "wpa_supplicant.conf" below are the files in the boot director of the PI3 with ASPBIAN STRETCH WITH DESKTOP that can connect to the wifi.

[image]

plankton99, 2017-10-14:
@ccrisan maybe you could send me your wpa_supplicant.conf  (with key inf xxx out) and I can try that one.?

jasaw, 2017-10-16:
@plankton99 This try wpa_supplicant.conf file. You'll need to:
1. Extract it from the ZIP file.
2. Change the country code.
3. Change the wifi SSID.
4. Change the wifi password.

[wpa_supplicant_example.zip](https://github.com/ccrisan/motioneyeos/files/1385949/wpa_supplicant_example.zip)

adrianrb101, 2017-10-26:
Thank you @RawBOT - I had a slight mis-type in my .conf file on the first boot, and although I fixed it before the second boot nothing I did would make it work.  Spent ages trying different editors, setting UNIX mode etc and nothing worked.  So frustrating.  Re-flashed the uSD card, saved the same .conf file, and it worked first time.  Why this OS doesn't re-read the .conf each boot is bizarre!

ccrisan, 2017-10-26:
@adrianrb101 because motionEyeOS will only read the wpa_supplicant.conf file from the boot partition **once**, at first boot. Then it will use the one on the data partition.

CapnWuwei, 2018-02-06:
Sorry if this is thread necromancy, I'm new here, and I'll cheerfully withdraw/move/etc.

I have the same problem. 

- with brand new pi 0 w, SD Card, Canakit Power Source, 
- 20180101 OS Image, utilizing jasaw's example wpa_supplicant.conf file and changes to ssid/password where necessary In Atom (with LF end of line), reimaging with each attempt to ensure changes for each first boot.
- after checking the process and troubleshooting in  [#804](https://github.com/ccrisan/motioneyeos/issues/804#issuecomment-286071303) ,  and [the troubelshooting guide](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems), 
- Raspian Stretch with Desktop OS (Kernel 4.9) was able to connect through GUI.
- router was restarted, and other devices were able to connect no problem.

This seems like an amazing program, I'd really like to get it working.

jasaw, 2018-02-06:
@CapnWuwei What's your SSID? Does it contain unicode? Are you able to attach your wpa_supplicant.conf file here, but change the sensitive information a little bit?

CapnWuwei, 2018-02-07:
@jasaw  sure! Here's whats on my wpa_supplicant.conf (with minor edits- all the Xs are something else  from A-Z and 0-9, letters are all caps). It's in the same folder as "bootcode.bin" "cmdline.txt" and "kernel.img"

I don't believe there's any special unicode characters, it's all capital letters, single digit #s, and  a period. I don't think the password is hex, it contains letters beyond F (very new to all this, please correct me when mistakes are made)

ctrl_interface=DIR=/var/run/wpa_supplicant
update_config=1
country=US

network={
        ssid="HXXE-4XXX-2.4"
        psk="AH3XXXXXXXXXPDF9"
}

wiggaworm, 2018-02-25:
All; I have pricked about with this for several hrs following all of the above, loaded this, loaded that and conclude it is £40 down the drain, Total Crap, obviously insufficient instructions, rubbish product. Fortunately I will use the PI Zero for something else ... (may be one more try when I am in a better mood but I wouldn't hold your breath)

jasaw, 2018-02-25:
@wiggaworm If you describe your problem, someone might be able to help you.

wiggaworm, 2018-02-26:
jasaw; thankyou for your prompt reply. My problem is exactly as bre75 commented on Apr 6 2017, top of this thread, I have done/checked everything here, Observed the first boot rule (doesn't help matters)  blahblahblah, unix format and everything spelt correctly. Done this dozens of times. No Joy. I have lots of PI's so its not as if I am new to this game, but on this one I give up.

wiggaworm, 2018-03-01:
OK peeps, I had another several goes; it seems \etc version of wpa_supplicant.conf (after first boot) does not contain same details as wpa_supplicant.conf file as left in "boot directory",  its missing ssid and psk details and presumably it should have these. I am probably on the right tracks now and could edit this back in on another PI platform but enough is enough, I really cant be bothered anymore ... all in all a poor show, with dreadful instructions its time to move on.

jasaw, 2018-03-02:
@ccrisan I'm wondering if we could improve the user experience by copying /boot/wpa_supplicant.conf file to /data/etc/wpa_supplicant.conf on each boot, provided source file exists. This would remove the need to reflash the SD card each time. I find myself running into this reflashing problem all the time during development/testing because I forgot to copy wpa_supplicant.conf before first boot.
I think the best user experience is to bring up a wifi access point if the wifi is not configured. Users can then configure the wifi via its web interface. Maybe let users disable wifi AP mode via web interface if user prefers ethernet interface.

@wiggaworm I'm interested in seeing what the differences are between your original /boot/wpa_supplicant.conf and /data/etc/wpa_supplicant.conf. Can you please attach both files here so we can debug the problem? Please change any sensitive information like SSID and psk before attaching the files.

ccrisan, 2018-03-03:
@jasaw I've been meaning to do something about this initial config on boot partition issue. The problem is  that those config files are often modified afterwards by motionEye or other utilities. Copying the boot variant over and over again at each boot would generate even more frustration.

A complex but viable solution would be to keep a hash of the boot variant and therefore be able to tell when it has changed. If the boot variant has changed, we use it to overwrite the data variant. And this mechanism could/should be applied for all config files that support "boot provisioning".

What do you think?

jasaw, 2018-03-04:
@ccrisan That's a good idea. Just have to be careful with a few scenarios:
1. Initial boot - copy wifi config from boot and save hash.
2. Wifi config changed via web interface - hash stays the same, do not copy from boot.
3. Updated wifi config in boot partition - hash is different, copy wifi config from boot and save hash.

In addition to your hash suggestion, what do you think about bringing up a wifi access point if wpa_supplicant.conf is not present in /data/etc and boot?

ccrisan, 2018-03-04:
@jasaw the AP stuff is still experimental and unreliable. I don't think automatically enabling it is a good idea, at least for now. Let's see how things go and we might think in the future of a *whitelist* (combination of board/wifi adapter) where this can be done, if wpa_supplicant.conf is not provided.