motioneye-project/motioneyeos #877: Turn on/off overall video surveillance gracefully

waqaslam, 2017-04-07:
Is there an easy and graceful way to turn on/off motioneye video surveillance? Like executing a command.

I've programmed an ESP8266 chip and hooked it up with a momentary-button. The plan is to use the button to turn on/off surveillance installed at my main door - as I don't want motioneye to record my movements when I'm home.

Currently I'm achieving this by simply starting/stoping the service:
```
sudo systemctl start motioneye       //to start
sudo systemctl stop motioneye        //to stop
```

But then I wonder, if there's any better way to switch this? Because sometimes it takes too long or fails to properly stop the service when motioneye is already busy in the process of recording.

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

closed 2017-06-01 by bortek

waqaslam, 2017-06-01:
Does motioneyeos has a forum? Couldn't find one.
Link please.

bortek, 2017-06-01:
https://groups.google.com/forum/#!forum/motioneye