motioneye-project/motioneyeos #878: Using 20170329 release, still not recognizing wifi on Pi Zero W

WhirlynerdHB, 2017-04-08:
I've checked out several threads on the W and still no luck--I'm probably missing a step as the only Pi I've setup before was a RuneAudio Pi 3.

I'm using this release:
motioneyeos-raspberrypi-20170329.img.gz flashed via Win32 disk imager.

I've added the wpa_supplicant.conf file to the root directory with my wifi info in it using Notepad++ on windows set to UNIX linefeed.

On boot the Pi doesn't see any wifi adapter--it attempts to configure a wired connection only.

Thanks for your patience and help!

RPiSpy, 2017-04-08:
In your wpa_supplicant.conf file are your SSID and password surrounded by double quotes? Have you triple checked they are correct? Double checking still wasn't enough for me :)

WhirlynerdHB, 2017-04-08:
Yup, I have the same issue sometimes!  They are double quotes but I'm thinking I'm not even getting that far--the boot screens I've seen have a 'configuring wireless network' line and I'm not getting that.  I DID triple check that my pi zero board has a W on it...

bortek, 2017-06-01:
I use the latest and it works. Try again.

closed 2017-06-01 by bortek