motioneye-project/motioneyeos #879: Mac OS X writeimage.sh makes bad wifi config file, unable to connect to wifi

robogeek, 2017-04-08:
I had first setup MotionEyeOS last summer using an old Raspberry Pi.  I got it working with a cheap WiFi dongle.  Today I wanted to update to the latest, and had trouble getting WiFi to work.  With the monitor and keyboard hooked up, I could see it was unable to connect to WiFi which caused a reboot.  Connect the ethernet cable, it boots correctly.  

Logging in I saw an error message in a log file in /data/var/log which I didn't record.  But it pointed me to this syntax error:

```
# cat /data/etc/wpa_supplicant.conf 
update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={
    scan_ssid=1
    ssid="WiFiNetworkName"
    psk="wifiPassword"
-e }
```

That `-e` on the final line is a syntax error, which caused the system to be unable to connect to WiFi.

The `writeimage.sh` script includes this:

```
# wifi
if [ -n "$SSID" ]; then
    msg "creating wireless configuration"
    conf=$BOOT/wpa_supplicant.conf
    echo "update_config=1" > $conf
    echo "ctrl_interface=/var/run/wpa_supplicant" >> $conf
    echo "network={" >> $conf
    echo "    scan_ssid=1" >> $conf
    echo "    ssid=\"$SSID\"" >> $conf
    if [ -n "$PSK" ]; then
        echo "    psk=\"$PSK\"" >> $conf
    fi
    echo -e "}\n" >> $conf
fi
```

Hurm, I thought that `-e` on that last `echo` command might be the culprit.

Checks:

1. Editing `/data/etc/wpa_supplicant.conf` to remove the `-e` made it so the system booted correctly.
1. Editing `writeimage.sh` to use `echo` rather than `echo -e` caused `/data/etc/wpa_supplicant.conf` to not have the syntax error and the system booted correctly

I see that on Linux, the `-e` argument enables interpolation of the `\n` sequence, and that on Mac OSX that argument is not available.

What I might suggest is to change that line to this:

```
    echo "}" >> $conf
    echo "" >> $conf
```

ccrisan, 2017-04-09:
That's very odd. I have no means to test this so I'll change it per your suggestion.

FritzTheCatfish, 2017-04-10:
This is "man echo" on OSX. There is no -e option
Please update

ECHO(1)                   BSD General Commands Manual                  ECHO(1)

NAME
     echo -- write arguments to the standard output

SYNOPSIS
     echo [-n] [string ...]

DESCRIPTION
     The echo utility writes any specified operands, separated by single blank (` ') characters and followed by a newline (`\n') character, to the standard output.

     The following option is available:

     -n    Do not print the trailing newline character.  This may also be achieved by appending `\c' to the end of the string, as is done by iBCS2 compatible systems.  Note that this option as well
           as the effect of `\c' are implementation-defined in IEEE Std 1003.1-2001 (``POSIX.1'') as amended by Cor. 1-2002.  Applications aiming for maximum portability are strongly encouraged to
           use printf(1) to suppress the newline character.

     Some shells may provide a builtin echo command which is similar or identical to this utility.  Most notably, the builtin echo in sh(1) does not accept the -n option.  Consult the builtin(1) man-
     ual page.

EXIT STATUS
     The echo utility exits 0 on success, and >0 if an error occurs.

SEE ALSO
     builtin(1), csh(1), printf(1), sh(1)

STANDARDS
     The echo utility conforms to IEEE Std 1003.1-2001 (``POSIX.1'') as amended by Cor. 1-2002.

BSD                             April 12, 2003                             BSD

ccrisan, 2017-05-07:
Fixed with 1624083dd8a2d5a4524792e21e8ee015dae30522.

closed 2017-05-07 by ccrisan