motioneye-project/motioneyeos #884: invalid date failed

burafili, 2017-04-12:
i download and burn my mmc with the 20170329 motioneyeos image.
looks work more good...
but i can't update the date that it rests at 1st jan 1970.
i remount / like w/r, i config new date.conf, save and reboot but 
nothing change. try http and ntp protocol... but nothing.

here my boot.log:

---- booting motionEyeOS 20170329----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting watchdog: done
 * Starting eudev: done
 * Setting root password: done
 * Configuring wired network: 192.168.1.22/24
 * Setting static gateway to 192.168.1.1: done
 * Setting static DNS server to 207.67.222.222: done
 * Setting current date using http: failed
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.22/24
 # Default gateway is 192.168.1.1
 # DNS server address is 207.67.222.222

any advise?

kop khun krap

ccrisan, 2017-04-12:
`Setting current date using http: failed` - your network configuration does not allow accessing Google's server (that is used to fetch the current date). Make sure your network configuration allows access on port 80 to google.com.

ccrisan, 2017-04-12:
Also make sure the DNS server resolves google.com correctly.

closed 2017-04-12 by ccrisan