motioneye-project/motioneyeos #886: streameye: not running, respawning

prometrik, 2017-04-12:
i installed motioneye on my raspberrypi b+ by following the directions at https://pimylifeup.com/raspberry-pi-security-camera/. when i start it up, i can access the pi through firefox on my windows 10 pc, but there is no video showing. it recognizes the camera and shows it. i have it enabled and it still is not showing video. when i plug in my pi to a monitor it says "streameye: not running, respawning" over and over and i can not put anything into the console to make it run. i am using this camera https://www.element14.com/community/community/raspberry-pi/raspberry-pi-accessories/raspberry-pi-camera-board. thanks for the help in advance

ccrisan, 2017-04-13:
1. What version of motionEyeOS is this? Please don't say "the last one".
2. Have you enabled Fast Network Camera?
3. Are you sure the camera is correctly connected and works as it should (i.e. have you tested it under Raspbian)?

prometrik, 2017-04-13:
sorry i meant to say the latest version. 20170329. its is on a b+ model so
i am running the version for a pi 1. i have the camera enabled it shows up
on my windows pc as there but there is no video or pictures coming through.
i then get the message i stated on the issue over and over. i did test it
under raspbian and it worked

ccrisan, 2017-04-13:
You never answered question number 2.

prometrik, 2017-04-13:
it is enabled

ccrisan, 2017-04-13:
So what is the problem again then? You first said "i have it enabled and it still is not showing video". Then you said "i have the camera enabled it shows up on my windows pc as there but there is no video or pictures coming through".

Can you please clearly state what your problem is? Some screenshots would help.

closed 2017-06-01 by bortek