motioneye-project/motioneyeos #887: Schedule system reboot

andrewvictor, 2017-04-13:
Hi, 

Am I able to run a script which reboots the system once every night?  If so, where can this be applied?

Thanks,
Andrew

ccrisan, 2017-04-13:
Use crontab: https://github.com/ccrisan/motioneyeos/wiki/Tweaks#running-commands-at-regular-intervals-cron-support.

closed 2017-04-13 by ccrisan

Naingenieur, 2018-04-25:
Hi,

First of all congratulation for all you have done with this project, I set 4 raspberry with it and it's really cool.

but I'm facing a problem.
I add a cron to reset my raspberrypis at some hour like this:
5 5,8,10,12,18,19,20,21 * * * * reboot

But it's seems that it doesn't reboot the raspberry because when I restart raspberry by SSH the camera setting was reset to have a correct picture but with cron the camera setting stay the same (too white during the day and too dark during the night)
But I have the reboot command only log in log file.
`Apr 25 20:00:00 camerarpi cron.info crond[818]: USER root pid 6539 cmd * reboot`
but with SSH I have all the start sequence.

Maybe I must set the cron command in another way?

I use crontab -e to edit cron.

Maybe I can restart only th camera process to reset this settings instead of restart all the raspberry? In that case, which process I need to restart?

Thanks for all help you could done.

ccrisan, 2018-04-25:
@Naingenieur don't you have one too many asterisks there in your crontab line?