motioneye-project/motioneyeos #888: Feature request: Privacy mask area

Lama81, 2017-04-13:
Hi,

Maybe a suggestion for a future update. Is it possible to add kprivacy mask setting. Then you can select and blackout some area. Somes surveillance solutions have this feature, but I don't know any that can do that on a raspberry pi. It will be a very cool add-on to MotioneyeOS.

Thanks !

Artens, 2017-04-15:
This should be possible, although it is not that easy as it would require a lot of additional software.
As motioneyeos is bases on Motion, please look into the documentation of Motion how to add a mask to ignore certain area(s) in the configuration.
It will not be as user friendly as the OS is now, but it is already possible if you really need it.

Motion can use a black and white picture (mask file) where the area is provided using the same size of picture. In the config file you provide the location of the mask file. Then, that area is ignored when you next time run Motion (or the OS, that is).

Your feature request would cover quite some coding I think, where you would need a page to load a picture, draw a mask (or multiple), save a mask file, update config file.

As already mentioned, the feature is already available in Motion, so you can help yourself in the mean time :)
I used it to block out a part of the street where lightbundles of cars would trigger a false positive for motion detection in the night.

Lama81, 2017-04-15:
Thanks for your answer. I was quite expect that getting that feature in5the GUI will require some work. I was thinking maybe it will be similar to the motion mask feature that is already available in the gui. 

 In the meantime, it's sure that I would take a look at the motion config that  you talk about. 

Thanks!

closed 2017-06-01 by bortek