motioneye-project/motioneyeos #890: Record both stills images and movies?

Jasonb11, 2017-04-15:
Is this a limitation of the OS or am I missing something?

The ultimate goal is to have the motioneyeos record both still images at an interval of 300 seconds. 

Still images setting:
[image]

Movies setting:
[image]

The problem is when I apply the setting for either the still images or movies to record at continuous or interval it overwrites the other setting and I can't use my desired setting. I'm thinking that I should find the file that applies to these settings and manually update the settings that way. 

Issue:
[image]
[image]

Currently, my settings are set to interval still images because I can create timelapse videos from them. But i would really like to have the continuous video too..

Any help or feedback would be most excellent, thanks!

ccrisan, 2017-04-16:
Unfortunately what you're trying to achieve is impossible using the available motion configuration options. That's why the UI automatically adapts the options for you. Sorry about that.

closed 2017-04-16 by ccrisan

Jasonb11, 2017-04-17:
:(