motioneye-project/motioneyeos #892: Logitech Webcam Image Funny

nagybal, 2017-04-16:
It is probably a linux/driver issue. But I do not know what to google for.
- funny colors
- funny resulution with little gray dots on the picture
[image]

please help

ccrisan, 2017-04-16:
I'll leave the issue open just in case other users know a remedy for this. Anyways, I think it's best if you clearly state your camera model.

davebdb, 2017-04-26:
What is the output of the dmesg at boot (or the snippet of dmesg.log containing the boot info), interested in knowing what Linux is detecting the camera as.  

Also what does the output of

/usr/bin/v4l2-ctl -d0 -L

Show for you?  if this is your only camera you can exclude the -d0 option, but if you have multiple cameras you need to figure out which device it is.  Had super-overexposure issues which I resolved and Calin was able to tell me what I needed to do to make my change persistent on reboots.  So if this is the only camera you just need to run

/usr/bin/v4l2-ctl -L

and copy/paste that output.  That should give all the camera settings you have set for the camera.  (even more than what is available on the WebUI [along with their ranges and defaults])

closed 2017-06-01 by bortek