motioneye-project/motioneyeos #893: Lots of errors on first boot Raspberry Pi 1

Toontje, 2017-04-16:
I get a lot of errors on first boot using a Raspberry 1B. It looks like the /data directory is not created on first boot.
[image]

Toontje, 2017-04-16:
Downloaded the 20170212 version as mentioned on https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices and this release doesn't have the issue.

ccrisan, 2017-04-16:
You *can* and you *should* use the latest prerelease. It's proven to be stable enough so far. Just rewrite it again and boot it. It sometimes happens with first boots to fail like that and it has to do with the data partition on the SD card.

closed 2017-04-16 by ccrisan