motioneye-project/motioneyeos #894: Updating dynamic DNS at DuckDNS with cron but without wget?

JohBod, 2017-04-16:
I managed to get pre-release 20170329 running with WiFi on my Pi Zero W at first attempt by writing a wpa_supplicant.conf file to the boot partition according to instructions at:

http://www.raspberrypi-spy.co.uk/2017/04/manually-setting-up-pi-wifi-using-wpa_supplicant-conf/

It works as advertised but now I want to add a cron job to update dynamic DNS at duckdns.org. The page:

https://github.com/ccrisan/motioneyeos/wiki/Tweaks#running-commands-at-regular-intervals-cron-support

gives an example of updating dynamic DNS at freedns.afraid.org using the wget command but I can't find wget in the 20170329 pre-release but the curl command is there, which I guess can be used for this puspose. I would be very grateful for some hint(s) how to use curl in a crontab line to update duckdns.

There are many alternatives for duckdns update through a simple http(s) call but the "no-parameter format" is probably the simplest:

https://duckdns.org/update/<YOURDOMAIN>/<YOURTOKEN>

How do I make this call with curl in a crontab line and how to I capture the response into a log file? ("OK" returned if success, "KO" if any failure).

Please excuse my ignorance, I am new to Linux...

Best Regards, Johan

ccrisan, 2017-04-16:
You're right. `wget` is gone and you should use `curl` instead. I have updated the wiki article and it now uses `curl`. As for your duckdns.org, just use `curl <your_url>` and you should be fine.

JohBod, 2017-04-17:
Many thanks Calin!

After some experimenting in the dark I realised I had to enclose the url and associated arguments to curl in double quotes ("...") to make it work as otherwise DuckDNS returned "KO" (error) instead of the wanted "OK" (success), maybe because something in the path stumbled on some odd character, perhaps the '&'. Below is my crontab line with domain prefix replaced by DDD... and secret token replaced by TTT...:

3,8,13,18,23,28,33,38,43,48,53,58 * * * * sleep 20 ; curl -s -S "https://www.duckdns.org/update?domains=DDDDDDDDD&token=TTTTTTT-TTTT-TTTT-TTTT-TTTTTTTTTTTT" >> /var/log/dyndns.log 2>&1 &

I googled around and found that curl argument -s suppressed the unwanted output from curl when redirecting output to a file (saves SD card space...) while -S (capital S) still allows error messages to pass through despite -s.

I think that I somewhat understand the crontab syntax, your example seems to run the job at hh:03 + n*5 minutes, but what does the sleep 20 do? My guess is that you want to run the wget/curl job 20 seconds after hh:03 + n*5 minutes to time-spread the load on the DDNS server (?) but the separating ';' will (according to my googling...) run the "sleep 20" command immediately in parallel with the curl command. && instead of ; would run the commands in sequence, I guess. Please correct me if I'm wrong.

Anyway, I can't complain, it works quite well now. The only problems left is to add a timestamp on each line in the dyndns.log file and to add newline 0x0A characters. Currently the "OK"/"KO" responses from DuckDNS appends to one huge single line in dyndns.log containing OKOKOKOKOKOKOKOK=....

What does the '&' at the end of the line do? I've got the impression that a trailing & on a shell command line makes the command run in the background while many sources say that *all* cron jobs run in the background anyway?

Thanks again. Your idea to combine all the cam' stuff with an OS into a self-contained black-box rpi image is simply great!

Please excuse my ignorance, but, as I said before, I'm a Linux newbie :-)

Best regards
Johan

ccrisan, 2017-04-18:
The sleep there was taken directly from the example given by https://freedns.afraid.org/. It's supposed to be a "random" number in order to "uniformly" distribute requests across users during those 5 minutes.

You are right about both `curl -s` and `curl -S`, as well as about the quotes. I have further updated the wiki article to reflect these changes, even if they aren't needed in the example's case.

closed 2017-04-18 by ccrisan

JohBod, 2017-04-18:
Thanks Calin,
    I did some googling and finally managed to write a crontab line
      that does exactly what I want, one line per update call with a
      timestamp followed by the response from DuckDNS. Like this:
    
    /var/log/dyndns.log:
    ...
    2017-04-18 23:43:43 OK
    2017-04-18 23:48:43 OK
    2017-04-18 23:53:43 OK
    2017-04-18 23:58:43 OK
    ...
    
    I noticed that you closed the #894 thread so I will not clutter up
    your "github issues" with this (since it's not an issue really) but
    in case someone asks, the crontab line looks like this:
    
    3,8,13,18,23,28,33,38,43,48,53,58 * * * * sleep 43 && printf
    "$(date "+%Y-%m-%d %T") " >> /var/log/dyndns.log &&
    curl -s -S
"https://www.duckdns.org/update?domains=XXXXXXXXXXX&token=XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX"
    >> /var/log/dyndns.log && printf "\n" >>
    /var/log/dyndns.log 2>&1 &
    
    Not sure if the trailing & is really needed but i doesn't seem
    to do any harm.
    
    Regards
    Johan
    
    Calin Crisan wrote:
    
    
      The sleep there was taken directly from the example given by https://freedns.afraid.org/.
        It's supposed to be a "random" number in order to "uniformly"
        distribute requests across users during those 5 minutes.
      You are right about both curl -s and curl
          -S, as well as about the quotes. I have further updated
        the wiki article to reflect these changes, even if they aren't
        needed in the example's case.
      —
          it on GitHub, or mute
          the thread.

Chiny91, 2017-04-19:
Perhaps it should be stated in this thread that a crontab does not require a trailing ampersand (&).   The command will run in the background when called.

motionEyeOS supports the _logger_ command, the standard way of putting your own timed/dated log item in the standard log file. `logger --help` will give the detail.    If you need to log your DuckDNS data to a DynDNS file (as above) then you would need to tackle syslog config, as well as logrotate (also needed for the printf method).   Note; I'm carefully not asking why you are getting motionEyeOS to do all this non-cam work that looks suitable for the router 😄

JohBod, 2017-04-19:
Thanks for the hints.
Looking at /etc/crontabs/root and /etc/logrotate.conf in MotioneyeOS, I draw the conclusion that logrotate operates on /var/log/*.log which includes my dyndns.log. Now that the update works OK, the log is no longer needed so I removed it.

Yes, the obvious choice is to let the router do the DDNS update but my ZTE junk router is hard to convince and my pi 0 w is the only computer on my LAN that currently needs DDNS.
I've just started to scratch the surface of Linux and have a _lot_ to learn... :-)