motioneye-project/motioneyeos #895: no boot on raspberry3 : can't open /dev/ttylogin

sol4ris2048, 2017-04-17:
Hello,

I'm trying to boot on my raspberrypi (with berryboot and squashfs img file) and it works perfectly with openelec and raspbian but with motioneyeos it does not work : i tried with released or prereleased versions (2017_03_26, 2017_03_29 and 2017_02_12) and at boot, the error is the same : "can't open /dev/ttylogin : no such file or directory"
And this message is on the loop for ever and ever......
Could you please help me? Thanks a lot....

ccrisan, 2017-04-17:
Have you tried one of the methods described [here](https://github.com/ccrisan/motioneyeos/wiki/Installation)?

sol4ris2048, 2017-04-17:
Tanks for your answer.  (I'm french so sorry for my bad english.....)

No i did not try this because i'm using berryboot and i downloaded
motioneye img file on berryboot website.

Then, because of the error i got, i took your 3 last released and i
transformed them into squashfs img files. And the error is still the
same.....

ccrisan, 2017-04-17:
Feel free to get back to us if you encounter problems using one of the recommended methods.

sol4ris2048, 2017-04-17:
Should i conclude that motioneyeos is not usable with berryboot?

ccrisan, 2017-04-17:
Honestly I don't know how berryboot works and judging by the errors you get, it probably does not work well with motionEyeOS.

agoldcheidt, 2017-05-27:
Hi @ccrisan it will be great if you can get motioneyeOS work under Berryboot, I run a website to download more than 170+ OS images for Berryboot and motioneyeOS is one of the most requested. I'm willing to do all tests on a Raspberry Pi 2 to make this work.

ccrisan, 2017-05-27:
What are the prerequisites for an OS image to work under Berryboot?

agoldcheidt, 2017-05-31:
Hi @ccrisan, the only thing that will not work under Berryboot is an OS image with a custom kernel or modules since Berryboot already include a kernel and default modules. If you check [these steps](http://www.berryterminal.com/doku.php/berryboot/adding_custom_distributions) you will have an idea on how Berryboot works. Basically, Berryboot compress the root partition via squashfs format and comment the all mounting points on the fstab file. That's it.

ccrisan, 2017-05-31:
motionEyeOS uses the Raspbian kernel with no change to the configuration. However the fstab is split in three different files and may pose problems.

I don't have the time to investigate this and I don't think this is in any way a priority, but if anyone wants to help with this, I'd be happy to take a look over their pull requests.

ccrisan, 2017-05-31:
Just to make it clear that porting motionEyeOS to Berryboot is still on the table, I'll reopen the issue.

agoldcheidt, 2017-06-01:
Alright, I've been checking and making some tests, right now I see some scripts on the /etc/init.d and I found that the file: "S01mountall" include the following function:

mk_tty_login() {
    test -z "$os_tty_login" && os_tty_login=tty1
    ln -sf /dev/$os_tty_login /dev/ttylogin
}

That function, from what I understand, make a symbolic link of /dev/$os_tty_login (I don't know what's the source for $os_tty_login) to /dev/ttylogin. I think this function is directly related to the issue: "can't open /dev/ttylogin : no such file or directory".

ccrisan, 2017-06-01:
The source for `$os_tty_login` is [os.conf](https://github.com/ccrisan/motioneyeos/blob/master/board/common/overlay/etc/os.conf).

Given that you have no `/dev/ttylogin` symlink on your setup indicates that the function `mk_tty_login()` is never called. The reason is most likely the fact that the previous [mount_fs](https://github.com/ccrisan/motioneyeos/blob/master/board/common/overlay/etc/init.d/S01mountall#L5) function fails. And it probably fails due to something messing up with the fstab system.

stevemg7, 2017-11-15:
I'd love to see this ported to Berryboot as well.  I tried converting the image to make it compatible for Berryboot and get the same error for the reason you say, about the fstab being split into 3 files.  I use my PI for several different OS's and so having it be on a dedicated memory card isn't an option....and everything else I use is able to be run from Berryboot....

ChristopherRush, 2018-09-16:
Still no update on this issue?

agoldcheidt, 2018-09-18:
Hi everyone, I'm sorry for haven't update this issue but yes, motion eyeOS is available to Berryboot thanks to an user who provide the fixes. You can download motion eyeOS here: http://berryboot.alexgoldcheidt.com/go/motion-eyeos