motioneye-project/motioneyeos #896: can't open /dev/ttylogin when booting up

Carson23, 2017-04-17:
[image]
I'm using a raspberry pi 3 B with the motioneyeos-raspberrypi3-20170212.img
on a Kingston 16gb sd card with an Ethernet connected to my network
I have no idea how to fix this

ccrisan, 2017-04-17:
Try rewriting the OS image from scratch. Something might have gone wrong at first write.

Carson23, 2017-04-17:
I have actually tried to rewrite it several times and the same thing popped up I may just be doing it wrong

Carson23, 2017-04-17:
I think it is worth mentioning that sometimes I get the reboot loop error when rewriting the card

Carson23, 2017-04-17:
[image]
These are the files that are on my SD card 
[image]
And does it matter which extract option I use when extracting to an img file?

Carson23, 2017-04-17:
Nvm I got it to work thanks for helping and responding quickly

closed 2017-04-17 by Carson23

ccrisan, 2017-04-18:
How did you fix the problem?

roelfsre, 2017-05-12:
I have the same issue. I have a PI Zero W, use the latest image and start from scratch. I only get to the point that it will ask for a login. But root, or admin does not work. No motioneye yet, looks like still booting, but ask for password.

Carson23, 2017-05-12:
I just tried it again and it worked

Carson23, 2017-05-12:
Do you have the right version?

Carson23, 2017-05-12:
Wait if it pops up login prompt you don't put in the password in the pi itself if you see an ip address type that in to your browser then when it asks for login type in admin as password