motioneye-project/motioneyeos #897: Motion not detected

FlyingPersian, 2017-04-17:
Hi
I just installed motionpie on my RPI B+. I'm currently running motionEyeOS 20170329, with motion version 4.0.1+git37b3595 and motionEye version 0.35.2. 

The problem is that motion is not detected and the stream is really slow and has a delay of about 6s. I set the frames for both the stream and video to 30fps. The resolution for videos is 1280x720. I hold and move my hand in front of the camera to test the motion and mostly it doesn't recognize it. If it does, the video can't be played (VLC). I can see my hand after about 5-6s in the stream, but it's not a video, rather single pictures every 5-10s. It's basically not smooth and lags out. 

I also haven't had pictures turned on, only videos, but it randomly created four really werid pictures. See here: http://imgur.com/a/niZKy

It's dark in my room right now, but I have the RPI NoIR cam and an IR-light. The picture I'm seeing in the stream is very good, no complaints there. 

I'm using Chrome 57.0.2987.133. 

Here the motion.log since I've installed motionpie: https://pastebin.com/xYtGiX4K
And here the motioneye.log since I've installed motionpie: https://pastebin.com/3jiGjxr3

ccrisan, 2017-04-18:
Your board (RPi +) is very slow and cannot handle everything you want it to do. Hence the lag. You'll need a more powerful board, such as a RPi3.

The weird picture could be because you have enabled *Create Debug Media Files* in *Motion Detection*.

FlyingPersian, 2017-04-18:
Oh okay. I saw tutorials doing that with a b+, so I figured it would work. Thanks! 

And yes, you are right, I did activate the debugging. Thanks! I'll try out on a better pi and get back if I still have issues.

closed 2017-04-18 by ccrisan

ari197, 2017-07-15:
Hi .. I just installed motioneye on RasPi2-B running with OSMC .. video streaming works fine .. but I can't get it to recognize any motion (it's not sending notification via pushbullet and not storing any image/video) ..

attached is my config ..

[thread-1.conf.txt](https://github.com/ccrisan/motioneyeos/files/1150064/thread-1.conf.txt)
[motion.conf.txt](https://github.com/ccrisan/motioneyeos/files/1150066/motion.conf.txt)
[motioneye.conf.txt](https://github.com/ccrisan/motioneyeos/files/1150065/motioneye.conf.txt)

ari197, 2017-07-15:
ok .. thisi is funny ... literally after I send the message, it works .. I don't know why though :(

FlyingPersian, 2017-12-20:
Hi
I upgraded to a RPI 3. Installed Raspbian, then motioneye according to the wiki. Everything seems to work fine, except motion detection. Tried different settings in the motion detection section. My current settings are "Frame Change Threshold: 0%", "Motion Gap: 1s", "Captured Before: 20f", "Captured After: 20f", and "Minimum Motion Frames: 1f". Capturing pictures manually works fine, motion triggered not.

motioneye.conf: https://hastebin.com/vaxewomota.vbs
motion.conf: https://hastebin.com/ahaponiloj.nginx
motion.conf in motioneye dir: https://hastebin.com/edokohobuq.rb 

Not sure, which motion.conf is being used, so I pasted both. Should be the first one though, since I turned on logging in the first motion.conf and I have the motion.log file in /var/log/

jasaw, 2017-12-20:
Not sure if it's related, but you've set your threshold to 0%. I haven't checked motion source code to see what 0 threshold mean. Can you test with a non-zero value?

FlyingPersian, 2017-12-20:
it was at 4% before. The less the number, the higher the motion sensitivity. Since the problem is on motioneye (not motioneyeos) I created a new issue here:

https://github.com/ccrisan/motioneye/issues/743