motioneye-project/motioneyeos #898: since now no subfolders on dropbox (and /CameraX)

CampoES, 2017-04-19:
Hi,

since some days my motioneyeos-System (Raspberry Pi 3+) creates no (daily)-subfolders on my Dropbox. I looked under /data/output/CameraX and there is no subfolder too. I'm sure that the system creates subfolders before.

Is there a possibility to fix this? How?

Thanks for your help.

Michael

ccrisan, 2017-04-20:
Can you post a screenshot of the settings panel, so I can see the values for your File Storage, Still Images and Movies sections?

Do media files actually get created somewhere, at all?

CampoES, 2017-04-20:
Hi ccrisan,

at the moment I only save still images which are stored under /data/output/Camera3 and on my dropbox in the right directory (/CampoES/), but not in subfolders (/CampoES/YYYY-MM-DD/*.jpg). 

Here are the screen-shots:

[image]

[image]

Michael

ccrisan, 2017-04-21:
Well yeah, you change the Image File Name option on the fly, using motion's internal web interface. And you set it to something that's not ideal for your use case: the filenames should have a pattern that makes them distinct. Not including the hour/minute/second information gives you one filename per day. I don't think that's what you were looking for.

CampoES, 2017-04-21:
I put the same configuration (Crontab and script) on another Raspberry. There it works fine. It makes snapshots (21-04-2017-snapshot.jpg, 21-04-2017-snapshot (1).jpg). On the raspberry the system overwrite the snapshotfiles, if more than one will be made on one day..Motion-images were saved under /2017-04-21/20-09-45-01.jpg. Before my selfinitiated alarm I had erased the folder /2017-04-21. At the alarm a new folder was created and the images were saved inside - everything as it should.

By the way: I only save one picture per day so the time is not essential for me, but for the system? I don't think so.

The error - so I think - is on another place.

closed 2017-06-01 by bortek