motioneye-project/motioneyeos #899: File Storage "Network Share" configuration/attributes

AshrafRezk, 2017-04-19:
[image]
**Network server**: the computer ip address obtained from IP config
**Share name**:       the shared folder on that computer
**Share username**: the computer administrator/account name
**Share password**: the computer login password
Q1: What is the root directory exactly?

ccrisan, 2017-04-20:
The Root Directory is the directory inside your share, where your media files will be uploaded. See https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#network-shares.

closed 2017-04-20 by ccrisan

starbasessd, 2020-10-06:
> I'm an elder newbie. I have followed the above meticulously, but
> constantly get "The network share failed:cannot mount Network share!" The
> share as I understand it, is my Mac on 192.186.1.xxx. on that Mac, on the
> desktop of that Mac I have created a file called FenceCam, where I wish the
> files to be stored. (The Raspi is, strangely enough, mounted on a fence in
> my garden and could easily be nicked by the people who smashed my security
> light, therefore requiring indoors storage that I can show to the
> magistrates after the theft of said Raspi cam). Therefore, as I understand
> it the Share Name is FenceCam. The Share Username is mine, all in lower
> case and no spaces and the share password is that which I use when booting
> up the Mac (A rare occasion as there is a Raspi2 sending out ADS-B
> information to aviation enthusiasts and a personal weather station sending
> information to the UK Met Office WoW program, amongst others. These are
> running 24/7) So now we come to Root Directory. My share is a folder, as
> described above, so how can it have a directory inside it? I would have
> thought that root would be the volume name inside the container disk In
> APFS, possibly something like /Macintosh HD/joebloggs/Desktop/FenceCam? I
> had also considered using the upload files system to save to my iCloud
> Drive- but Uh Oh, no iCloud Drive****

How are you sharing the folder on the MAC's desktop to the network?
NFS, SMB/SAMBA/CIFS, FTP? Something else? Folders are not shred by default.
I have an older iMAC that I can try to set up the same type share...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-06:
> Kevin, thank you for such a speedy reply. I have, I believe,
> enabled SMB
>
> A342

I set a share on my desktop called 'test'
In a PiZero, I set up File Storage, Network Share
Network Server: IP Address of my iMac (set to a static IP, best for any
device that shares info)
SMB Protocol: 1.0
Share Name: test
Share Username: <my iMac user>
Share Password: <my iMac Password>
Root Directory: /
That will then allow any other added folders under there to work
If you didn't change anything under Movies, it default created folders by
Year Month & Day, then file names for each movie.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

McGinty342, 2020-10-06:
Kevin, 
         Thank you so very much. I think I was getting blindsided by the use of “share” rather than folder and directories within folders. I am now saving the files on my Mac and when the raps goes walkabout I will have the evidence required. 
Again, many thanks and hopefully this will have benefitted others in the community as well.

A342

starbasessd, 2020-10-06:
Glad to help.Hope you get your perpetrators.

On Tue, Oct 6, 2020 at 1:50 PM McGinty342 <notifications@github.com> wrote:

> Kevin,
> Thank you so very much. I think I was getting blindsided by the use of
> “share” rather than folder and directories within folders. I am now saving
> the files on my Mac and when the raps goes [...]
-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

McGinty342, 2021-02-20:
Kevin, greetings. I trust Covid has not entered your door. You provided me with the solution to getting my single Raspi Zero W up and running with  MotiomEye OS. I seek your help in that twice now the system has stopped recognising my network share (and thereby the camera) for no good reason. I have changed nothing, and the system is not connected to the outside world. It has occasionally stopped but a reboot via recycling the power always got it back online. The camera is working, as seen by changing the storage from network share. I have studied dozens of similar cases on GitHub but can make neither head nor tail of any of the proposed solutions nor indeed of the language used, hence a call to you to see if you have any knowledge of or fix for the problem.

 As a courtesy in return for yours (As a glider pilot/instructor (retired) I’m an avid map reader)  54° 38’ 40” N 05° 38’ 38” W

starbasessd, 2021-02-20:
With any 'intermittent' issue on a Pi that goes away (for at least a while)
the first thing I look at are heat and power to the Pi.
You can look at the messages.log for thermal (it posts the CPU temp
regularly) for a start.
If you can ssh or console into the Pi, you can run the command
vcgencmd get_throttled
Which will give you 0x0 if it hasn't had any issues, or a very large number
which you can look up to see which area is giving trouble since the last
reboot.
Let me know if you need more specific instructions to view this information.
I visited GB (London Region) a couple of years back. Ireland & Scotland are
still on my bucket list. I'm a little predjudiced about the BlueGrass
around here, but from other family visitors reports, you're in a very
lovely area, too.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?