motioneye-project/motioneyeos #900: Wifi usb dongle (rpi_2)

AshrafRezk, 2017-04-19:
Any clue why a wifi dongle might not even light up in order to work?
History of my project: the wifi dongle used to work beautifully... now out of the blue, its led doesn't even light up in the boot; hence, no wifi connection is achieved.

ccrisan, 2017-04-21:
* see https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems
* does it work with Raspbian?
* attach your boot.log

cristimiha, 2017-04-23:
Calin, I searched all the documentation and i cannot find a procedure to connect using a wifi connection.
I know that this is below the noob level, but still :)

AshrafRezk, 2017-05-01:
Hey,
Thank you!!
I resolved the issue by simply reburning the OS on the sd card. It has been working like a charm for days :)

AshrafRezk, 2017-05-01:
@cristimiha 
All you need to do is to plug in the Wifi dongle (in case of not having rpi3)
such as: https://www.raspberrypi.org/products/usb-wifi-dongle/
Making sure that it is compatible.
.....
Once it's plugged into the rpi board via usb.
You go in the network tab and enable wireless network.
Note: the rpi will act as a client in an already existent network NOT a host/hotspot.
You type in the wireless SSID/name that is in range accurately and its corresponding password.

closed 2017-05-18 by silversword411