motioneye-project/motioneyeos #906: No Menus available

mediastore93, 2017-04-21:
Everything was working fine on my Pi Zero W, I changed some of the settings (increased framerate etc) then next time I open the webpage I only see the camera image (which is streaming fine) and no menu panels. The webpage doesn't fully load (which is why I suspect I can't see the menus) but I can't change the setting to make the streaming image smaller as I suspect the stream is using too much bandwith. Can I change the settings by ssh'ing into a config file?

ccrisan, 2017-04-22:
Screenshot?

mediastore93, 2017-04-22:
Thanks for your help with this!

[grab1.pdf](https://github.com/ccrisan/motioneyeos/files/949061/grab1.pdf)

mediastore93, 2017-04-22:
And the page never fully loads up. I've rebooted, tried different browsers etc.

ccrisan, 2017-04-22:
Well the pdf doesn't say too much. What is the URL that you access when you open the UI?

mediastore93, 2017-04-23:
I can watch the stream with 198.168.0.10:8080 etc in the browser fine just cant see any menu buttons to change settings.

ccrisan, 2017-04-23:
Don't use port 8080 - that's for streaming only. Use port 80 (i.e. no port).

closed 2017-04-23 by ccrisan