motioneye-project/motioneyeos #907: Compatible with NanoPi

escoand, 2017-04-22:
Is there a release for the NanoPi family, espacially for the NanoPi NEO Air?
http://nanopi.org/NanoPi-NEO-air_Feature.html

ccrisan, 2017-04-22:
Nope, there's no NanoPi support. For your next question, see https://github.com/ccrisan/motioneyeos/wiki/FAQ#can-motioneyeos-be-ported-to-insert-your-board-here.

escoand, 2017-04-22:
Ok, I understand this. Do you have a short todo list what needs to be ported?

ccrisan, 2017-04-22:
I'm currently waiting for a PINE A64 board which looks promising and which I hope will run motionEyeOS soon.

escoand, 2017-04-22:
Kernel is probably the most important thing.

ccrisan, 2017-04-22:
Indeed, once the boot loader (usually U-Boot) and the kernel work together, everything becomes piece of cake.

escoand, 2017-04-23:
Ok, i'll have a look.

Off-topic, my reason for using NanoPi was the size and the huge amount of cheap actioncam outdoor cases. Usually too small for RaspiZero. Do you know a good alternative?

closed 2017-06-01 by bortek