motioneye-project/motioneyeos #908: odroid-xu4 doesn't boot

louis141, 2017-04-22:
Hi,
my odroid-xu4 doesn't boot since version 20170326. On the sreen is only the blinking coursor.
The blue led on the device is blinking too. 
Version 20170212 works very well.
What can I do? Thx.

ccrisan, 2017-04-22:
I myself use an XU4 with 20170326. Did you try rewriting the image from scratch? Is this issue of yours reproducible?

louis141, 2017-04-23:
Yes, from scratch. (SDFormatter with option fomat size adjustment. Then burned with Win32DiskImager. 100% reproducible. 
BUT: these behavior was with my eMMC storage card.
Now I tried again with a MicroSD card (20170329). It is working!
It seems like a problem since 20170326 with boot from eMMC storage card.

ccrisan, 2017-04-23:
To be honest I never tried the emmc storage as I use simple SD cards to test motionEyeOS. I don't have one at my disposal, but I believe the device path has changed with the new kernel (4.9, which is used by the latest version of motionEyeOS).

I'd suggest trying the following changes to see if the emmc path is now `/dev/mmcblk0` (as I presume):
 * in `<boot partition>/boot.ini`, replace `mmcblk1p2` with `mmcblk0p2`
 * in `<root partition>/etc/fstab.disk`, replace `mmcblk1p` with `mmcblk0p`

You'll need a Linux box (or OSX/Windows with ext4 drivers) to make these changes. If they really work, keep in mind that the changes would be overwritten by any subsequent firmware update.

louis141, 2017-04-23:
Yeah! Now it works!!!
Thank's a lot.
Great job. :)

ccrisan, 2017-04-24:
I won't close the issue just yet, as this is not a proper fix. I'll think of a way to add proper support for emmc on the xu4.

DaveBF, 2017-09-22:
Hello,
I had the same problem booting 20170827 until I made the same changes.  That allowed the board to boot, but I'm getting lots of "No such file or directory" errors.  Are there more changes that need to be made to the fstab.disk or other files maybe?
Thanks!

ccrisan, 2017-09-22:
@DaveBF in theory no, but I'd need to see some context for those `No such file or directory` errors. Where do you see them? Logs, screenshots, pictures are welcomed :)

DaveBF, 2017-09-25:
Well, i cant get into the console, so i wont be able to get to log.  SSH isnt working either.  So, how about I transcribe some errors?  There will be typos.  

```
* Detecting disk device: /dev/mmcblk0
* Mounting filesystem: mount: mounting /dev/mmcblk1p3 on /data failed: No such file or directory failed  
cp: can't create '/data/etc/os.conf: No such file or directory /etc/init.d/conf: line 15 /data/etc/os.conf: No such file or directory
* Loading kernel modules: done
* Setting hostname: done
* Starting syslogd: /etc/init.d/S04syslog: line 12: /var/log/dmesg.log: No such file or directory done /etc/init.d/S04syslog: line 13: /var/log/dmesg.log: No such file for directory
* Starting endev: done
* Starting watchdog: done
* Post-upgrading to version 20170827: failed
<date string> user.notice post upgrade: mkdir: cant create directory '/data/varlib': Read only file system
<date string> user.notice post upgrade: /usr/share/post-upgrade/20170827.sh: line 7: /data/etc/motioneye.conf: No such file or directory cp: cant create '/data/etc/version': No such file or directory
```

Well, I think you get the idea... lots of No such file or directory errors until the login prompt appears (which doesn't let me login).

In the second message above, does the mmcblk1p3 need to be changed to mmcblk0p3? in the config files?

thanks,
Dave

ccrisan, 2017-09-26:
mmcblk1 is for emmc, while mmcblk0 is for regular sd card.

DaveBF, 2017-09-26:
In your Apr 23 post, you told louis141 to change from mmcblk1 to mmcblk0 to get the emmc card to boot.  He did that and it was successful and I did that and it was partially successful.
The emmc card does NOT boot with mmcblk1.  All I get is a flashing cursor.

ccrisan, 2017-09-26:
Things have changed between kernel 3.x and 4.x. I have no idea why. Please try with both and see which one works.

DaveBF, 2017-09-26:
I have tried both.  mmcblk1 does not work at all.  mmcblk0 boots but then fails when trying to mount file systems.  When I get some time, ill go make some additional changes to the config files and see what happens.

ccrisan, 2017-09-27:
@DaveBF then I'm out of ideas. I don't own such a card and I couldn't possibly test this. I was relying on what other user(s) have reported.

DaveBF, 2017-09-30:
Ok, I finally got it to work.  Both lines in `fstab.disk` (`/boot` and `/data`) need to be changed from `mmcblk1` to `mmcblk0`.  Either that wasnt obvious to me or I just missed it.  Its up and running now on the eMMC module.

ccrisan, 2017-09-30:
@DaveBF thanks for the donation. I'll order an EMMC (with adapter :P) soon as I've been meaning to do so for some time now, out of curiosity.

Yeap, both lines need to be adjusted. See my comment from above:

>  * in `<boot partition>/boot.ini`, replace `mmcblk1p2` with `mmcblk0p2`
>  * in `<root partition>/etc/fstab.disk`, replace `mmcblk1p` with `mmcblk0p`

Did you also replace it in `<boot partition>/boot.ini`? The biggest problem with this setup is that the boot partition will be overwritten by subsequent updates and it won't work anymore after an upgrade. At least not without another manual change. There's no easy way that I can think of to have this fixed.