motioneye-project/motioneyeos #911: Google Drive Upload Issues

sheppy1, 2017-04-23:
Hi :)

Having issues here with Google Drive uploads, it used to work but now suddenly stopped. Attached is my motioneye log file.

Any ideas?

Thanks
[motioneye (2).txt](https://github.com/ccrisan/motioneyeos/files/949902/motioneye.2.txt)

ccrisan, 2017-04-23:
You have a lot of `MemoryError`s in your log file. Upgrade to a board with more memory or clean up your media files as browsing them is not as memory-efficient as it should...

sheppy1, 2017-04-23:
I don't think there are any Pis with more memory are there? I have the Pi 3. And there isn't even 1 days worth of files on there yet... It's only been up and running for about 18 hours.. So nothing really to clean up

ccrisan, 2017-04-23:
I see. In that case, I'd run `top` to see who eats up that much memory. Do you have relatively large movie files? They could eat up that amount of memory when loaded into RAM.

sheppy1, 2017-04-23:
Goof suggestion! I've attached a top output, does anything look amiss to you? Not being lazy, I'm just not clued up on the ins and outs of Linux! I do appreciate your help too! :)
[image]

Also the files do appear to be larger than expected! Around 400mb for a 5 min clip! Not sure why as I haven't maxed out any of the quality settings. Are there any I can/should adjust to cut down the size of the movie files?

Thanks!

ccrisan, 2017-04-24:
Indeed, the python interpreter eats up a lot of memory. You'll need to tweak your movie settings a bit. Run some tests with various options (codec, quality, resolution, framerate) and see which offers the best compromise.

sheppy1, 2017-04-24:
Great! Thanks for that :) I changed from .mp4 to .flv which brought the average 5 minute clip down from about 300mb (which seems awful large!) to about 123mb :) and this seems to have resolved the upload issue! :)

Final question and sorry for pestering! Just easier to ask now than raise another post! Is it possible to skip the local upload and have every clip go directly to Google Drive?

Thanks

ccrisan, 2017-04-25:
Unfortunately it's not possible to upload the movies directly to Google Drive. They have to be created somewhere first by the motion daemon. Now they could be stored in memory with some workarounds (tmpfs) but that could have other negative impacts on the memory usage.

closed 2017-04-25 by ccrisan