motioneye-project/motioneyeos #912: motioneye OS not start

bmili, 2017-04-23:
Hi!

I've been running MotioneyeOS  for month, but after it has been shut down, I couldn't run it again.  It's restarting all the time. 

http://allmediafiles.s3.amazonaws.com/20170423_230652a.jpg

regards
bmil

ccrisan, 2017-04-24:
`No space left on device` means that you'll basically need to remove some media files (pictures, movies) from your data partition. For that, you'll need a Linux machine (or an OSX/Windows with ext4 filesystem support).

closed 2017-04-24 by ccrisan

bmili, 2017-04-24:
Hi!

I can't mount my sd card to my linux, because  is running in cloud!

Can I  access my data partition from Ext2Fsd ? It is a Windows file system
driver for the Ext2, Ext3, and Ext4 file systems. It allows Windows to read
Linux file systems natively, providing access to the file system via a
drive letter that any program can access.
Thank you!

Regards

Boris

ccrisan, 2017-04-25:
@bmili normally yes. I remember using something like ext2fsd back in the days when... I had no other choice. Just make sure you have write support :)

bmili, 2017-04-25:
Done!

Message file was almost 30Gib on 32Gib SD card.

Thank you!

Regrads

Boris

ccrisan, 2017-04-25:
@bmili what version is this? Newer versions will automatically rotate those log files.

bmili, 2017-06-25:
>
> H!

Why I can't run motion eye  (installed manualy
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) localy?

http://127.0.0.1:8081/

ERR_CONNECTION_REFUSED

 Boris

 [image: Facebook] <http://www.facebook.com/borismil> [image: Twitter]
<http://www.twitter.com/borismi>
http://www.winsum.ws