motioneye-project/motioneyeos #915: MotionEyeOS camera menu item request

davebdb, 2017-04-25:
I recently purchased a GEARHEAD 4MP 720P HD WEBCAM and plugged it into the Raspberry Pi to run on MotionEyeOS.  Now having it set to monitor the outside, I came across an issue where it was overexposing.  So I decided to troubleshoot myself and find out the different v4l2-ctl commands to issue and ran them individually trying different settings out.  The item that worked was the 'power_line_frequency' setting which is an integer value (at least for this camera).  I tried setting up a startup script within the /etc/init.d/ but that is a read-only file system and I really don't want to mess around with the init.d filesystem at all.  So I was wondering if it is possible to configure this item in the WebUI for MEOS.  

Also because I have multiple cameras it helps to define which device.

So my entry to make my system actually go with proper exposure was

/usr/bin/v4l2-ctl -d 0 --set-ctrl power_line_frequency=0

On my other camera it is set to 3 is auto.  Below I am going to paste the output of both cameras values for your reference.   Device 0 is the new camera (gearhead) and Device 1 is another camera, can't remember model off hand.

[root@PiCam01 init.d]# /usr/bin/v4l2-ctl -d0 -L
                     brightness (int)    : min=-64 max=64 step=1 default=-8193 value=16
                       contrast (int)    : min=0 max=64 step=1 default=57343 value=38
                     saturation (int)    : min=1 max=128 step=1 default=57343 value=77
                            hue (int)    : min=-40 max=40 step=1 default=-8193 value=-2
                          gamma (int)    : min=72 max=500 step=1 default=57343 value=100
                           gain (int)    : min=0 max=100 step=1 default=57343 value=0
           power_line_frequency (menu)   : min=0 max=2 default=1 value=0
                                0: Disabled
                                1: 50 Hz
                                2: 60 Hz
                      sharpness (int)    : min=0 max=6 step=1 default=57343 value=3
         backlight_compensation (int)    : min=0 max=2 step=1 default=57343 value=1
[root@PiCam01 init.d]# /usr/bin/v4l2-ctl -d1 -L

User Controls

                     brightness (int)    : min=0 max=255 step=1 default=127 value=139 flags=slider
                       contrast (int)    : min=0 max=255 step=1 default=127 value=152 flags=slider
                horizontal_flip (bool)   : default=0 value=0
                  vertical_flip (bool)   : default=0 value=0
           power_line_frequency (menu)   : min=0 max=3 default=3 value=3
                                0: Disabled
                                1: 50 Hz
                                2: 60 Hz
                                3: Auto
[root@PiCam01 init.d]#

So found the problem and the resolution, but no way of setting it to save as it is a CLI change that is lost on reboot.

ccrisan, 2017-04-26:
See https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-custom-startup-commands.

davebdb, 2017-04-26:
Thank you Calin. That worked and since it is a shell script, I can add comments with standard # prefix to the line.  So until a feature like what I mentioned is available, this is a good substitute.

-EOF-

closed 2017-04-26 by davebdb