motioneye-project/motioneyeos #916: motion eye internet setting

lbin199012, 2017-04-26:
I set up the Virtual Servers / Port Forwarding for my raspberry pi camera. I can access the camera through htttp://xx.xx.xxx.xxx.8081 . However, there is not any setting panel. I only can view, but can not do anything. It is not same with view through 192.168.0.x in the local network. How can I do to make htttp://xx.xx.xxx.xxx.8081 same as 192.168.0.x in the local network? In 192.168.0.x I can change setting, view video record and so on.

bscuderi13, 2017-04-26:
Forward port 80 instead if you want the whole menus and program.

silversword411, 2017-05-17:
Answered.

With the right router, you can also change ports in the forwarding rule so external is one port, internal is another.

Closing ticket

closed 2017-05-17 by silversword411