motioneye-project/motioneyeos #919: Removing the spinny between history images

elyobelyob, 2017-04-27:
The spinny thing takes 10% of the screen real estate, it's a bit annoying when zipping through photos. I'd prefer something less central and more invisible or out of the eye line. 

Any suggestions on how to achieve that?

Thanks

Chiny91, 2017-04-27:
"The spinny thing"; this ?
[image]
It takes up less than 1% of my screen, although agreed it is in the centre.   Or something else produced by your computer/laptop/tablet/phone ?   That spinny lasts a fair while downloading the entire picture library but less than 1 second in between pics so hard to say a problem.

Depending on your habits/preference, it can be easier to transfer (I use rsync as a cron job) the pictures to a faster box, leaving motionEyeOS to get on with the business of watching the world.   No delays, no spinny.

elyobelyob, 2017-04-27:
That's the thing. I tend to go through my photos by holding down on the left/right arrow to scroll through my photos, only stopping when one looks interesting. This basically means the spinny thing is there pretty much permanently when scrolling through the hundreds of images. Not a show stopper, just thought I'd raise it as a thing.

elyobelyob, 2017-04-27:
Here's an example of the spinny thing ... https://youtu.be/Al1_UNurEIg

elyobelyob, 2017-04-27:
And he wasn't stealing my motorbike, just looking

elyobelyob, 2017-04-27:
To be fair, I'd love if the left & right buttons faded out as well.

A very reliable bit of software otherwise. I love it.

Chiny91, 2017-04-28:
OK, I see what you mean.   I think what you must be doing is keeping your finger on one of the back/forward buttons, essentially producing a video.   Your R Pi or whatever must be straining its cpu with all that effort 😄 and hence the spinny.   For Youtube, you probably would be better off actually producing a video - no spinny then.

Is your camera misting slightly ?   Dusty ?

elyobelyob, 2017-04-28:
I put that up there as an example of flicking through the photos. I don't make videos to put on youtube. If I was, I'd download the original images and create that using video software.

So, I'll review my photos, yes, holding onto a button to zip through quickly. I just think it'd be nice to have the option to fade out those whilst doing that.

The camera is poking through a hole, which I need to fix as it's slipped a little.

closed 2017-06-01 by bortek