motioneye-project/motioneyeos #920: Raspberry Pi Camera Module V2 8Mp with 90 fpm

hyolux, 2017-04-27:
Hello everyone,

I'm very new to this, I have one issue and I would be happy to get some help.
I would use the Raspberry Pi Camera Module V2 8Mp with a resolution of 640 x 480 and 90 frames per second with motioneyeos.

I'm not sure if it is technically possible with motioneyeos. I saw that the v4l2 driver also support these settings. The motioneyeos webinterface only offers 30 fpm maximum. I tried to change the framerate in the _thread.conf_ file. I changed the _stream_maxrate_ value to 90 and the _framerate_ value to 90.
The webinterface shows now 30 fps, but I'm not sure if this is sufficient? It seems that the camera is not
showing the livevideo with 90 fpm.

Am I right or do I need to modify more than these settings?

Thanks for any help in advance :-)

redflag237, 2017-05-02:
Using 8MP you're creating a meaningful load for your RasPi CPU. @30fps i think you hit the maximum data rate the SPI Interface can handle.

hyolux, 2017-05-04:
Hi redflag237,
I don't have any experience concerning the performance of the RasPi and high camera framerates.
I think it would be the best to try it out and see :-)
Actually the system runs fine, but I'm not sure that the live image is really 640 x 480 @ 90 fps?

closed 2017-06-01 by bortek