motioneye-project/motioneyeos #922: Raspberry Pi offline

gigawert, 2017-04-29:
I can't access my Raspberry Pi with motioneyeOS installed from a different PC on the network. I have not accessed it in a while but I do remember looking at the page a month or so ago. How can I do it now?

redflag237, 2017-04-30:
rebooted?

gigawert, 2017-05-02:
Yes, still not working

redflag237, 2017-05-02:
Why don't you plug in a keyboard and a screen. Then check network connectivity from the terminal.
`ifconfig` or `iwconfig` might bring you the needed information.

gigawert, 2017-05-03:
Oh, by looking at the console it said that it was expecting ethernet. Easy fix, I just plugged it into ethernet, accessed the page from another computer on the network, then configured the wireless network.

closed 2017-06-01 by bortek