motioneye-project/motioneyeos #923: No network share available

snowman82, 2017-04-30:
Hi,

I dont have option to save files in a network share. Is there any option I should enable ? 

[image]

Thanks,

ccrisan, 2017-05-01:
Are you sure this is motionEyeOS and not just motionEye? What version is this?

snowman82, 2017-05-01:
Indeed it is. Posted in the wrong forum? Can you please point me to the right one?

Thanks

ccrisan, 2017-05-01:
Well, motionEye lives here: https://github.com/ccrisan/motioneye. However if you want network shares, you just need to enable [smb_shares](https://github.com/ccrisan/motioneye/wiki/Configuration-File#smb_shares) and make sure you run motionEye as root.

closed 2017-05-01 by ccrisan