motioneye-project/motioneyeos #928: wpa_supplicant secure and more then one

MatzeP, 2017-05-05:
Hi,
I know that my questions are possible in Linux but is it also possible in motioneyeOS?

is it possible to create a wpa-file were the psk isn't in clear text?
Like "wpa_passphrase MYSSID passphrase" ?

Can I configure more then one SSID in the File?

ccrisan, 2017-05-05:
The wpa_supplicant daemon as well as the syntax for its configuration file are the same as those included with any recent Linux distro. This being said, whatever conf works for you on a regular distro should work on motionEyeOS.

The only problem I see is that your config file might be overwritten by motionEye in a way that breaks it, depending on what options you use. If you find out that motionEye breaks your wpa_supplicant.conf file, just disable motionEye's WiFi management by removing/commenting out the line `wpa_supplicant_conf /data/etc/wpa_supplicant.conf` in `/data/etc/motioneye.conf`.

MatzeP, 2017-05-05:
Hi,
If I want use wpa_passphrase then I must do your explainaition to switch off the only read mode. correct?

I asked this, because I configure the device in the one WLAN and take away to another WLAN were the device work and the device should connect on both WLAN.

silversword411, 2017-05-17:
I believe question was answered. Full configuration into for wpa_supplicant is here for further reference:
https://wiki.archlinux.org/index.php/WPA_supplicant

Closing ticket.

closed 2017-05-17 by silversword411