motioneye-project/motioneyeos #930: motioneyeos on raspberry pi zero : infinite reboot sequence

VincentGaudeul, 2017-05-06:
Hi. I'm facing problems starting wifi on my raspberry pi zero device, with latest motioneyeos distribution. 

I've got a running wpa_supplicant.conf configuration file in the boot partition. This file has been tested ok on latest raspbian distribution. Wifi was ok. 

With motioneyeos, I have the following message : 

Starting wpa_supplicant : no device
Configuring wireless network: no device

and then infinite boot sequence. 

I remembered I had to make some changes in /etc/network/interfaces file on raspbian, to make it work. Is it a possible issue ? 

Thanks for your help, and for your great work by the way :-)

ccrisan, 2017-05-06:
Is this a Zero W or just a plain Zero board?

VincentGaudeul, 2017-05-06:
It is (of course) a zero W board. 
As I said, the wifi was running properly on the same zero W board hosting a microSD card with latest raspbian distribution.

ccrisan, 2017-05-06:
And what motionEyeOS version do you use? Please don't say "the latest".

VincentGaudeul, 2017-05-06:
motioneyeos-raspberrypi-20170212

ccrisan, 2017-05-06:
Use 20170329. 20170212 doesn't support Zero W.

closed 2017-05-06 by ccrisan

sheppy1, 2017-05-06:
Now while I don't understand your error message I I did get infinite reboots on my Pi 3 and it was down to the USB charger I was using. I changed to a different one and all was fine after that.

VincentGaudeul, 2017-05-06:
Thanks. I'll try 20170329 version. 
For information, this version does not appear in https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices, does it ?

ccrisan, 2017-05-06:
@VincentGaudeul it's still a prerelease and I have yet to release a stable version. The idea is to only list stable versions in the Supported Devices page.

VincentGaudeul, 2017-05-06:
Great. It works now, as a charm :)
Thanks for this extraordinary quick fix !!!!!!!!!!

sriniiyer, 2017-11-09:
Hi guys,

I'm having this exact same issue with the latest version (20171008) as well as with 20170329. Any suggestions? I'm using the pi zero w.

Thanks!

jasaw, 2017-11-09:
@sriniiyer Have you configured wifi credentials on your Zero W? https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration
The device will keep rebooting when there's no network connection.

sriniiyer, 2017-11-09:
Yes, I have: This is what my wpa_supplicant.conf contains

```

country=us
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
 scan_ssid=1
 ssid="Hogwarts"
 psk="mypassword"
}

```

Just FYI, the first time I booted it, it worked and I was able to connect to it through a browser. After that, it just started going into this reboot loop.

Ive reflashed the SD card and copied the wpa_supplicant file again, but it still never starts up. I'm not sure what is wrong. Is my file format wrong? Do extra spaces matter?

jasaw, 2017-11-09:
@sriniiyer You managed to connect to it via a web browser, that means the wpa_supplicant.conf file is correct. Could it be a wifi signal strength issue? Have you tried moving the Zero W next to your wifi access point to rule that out? Another thing to consider is your power supply and USB cable.

sriniiyer, 2017-11-09:
Yes, I moved it just next to my router and had the same issue. 

I'm using the official power supply from adafruit, but can try changing it, but it worked the first time, so im wondering if that is actually the problem. 

Is there any chance that my wifi hardware burnt out ? Its the first time that I'm using the pi. 

Should I check if the wpa_supplicant.conf has been copied to /etc/ or not? Maybe I can mount the filesystem and check that from my laptop?

jasaw, 2017-11-09:
I believe the wpa_supplicant.conf file is copied to /data/etc/ directory. No harm checking that file to make sure it hasn't changed.

It's quite unlikely that the hardware is damaged, unless you accidentally zapped it with ESD.